d-ronin/dRonin #6: Branding

mlyle, 2015-10-04:
About box, license file, logo.

Blocks on getting logo from fiverr.

tgreer, 2015-11-25:
Ref #30

tgreer, 2015-11-25:
Logo files which are referenced throughout the code are:
taulabs_logo_128.png
taulabs_logo_256.png
taulabs_logo_32.png
taulabs_logo_64.png
welcome-logo.png
tau_trans.png

fujin, 2015-11-25:
@tgreer thx, added to issue

tracernz, 2015-11-26:
I will fix the splash screen up to make it look nicer too, I have a branch somewhere that does that to Tau (it seems to be anti-alised etc. already on OSX, but not Linux).

tgreer, 2015-11-26:
#58 is a prerequisite of this

tracernz, 2015-12-04:
* [ ] ground/gcs/src/plugins/config/images/TX2.png

VArcht, 2015-12-04:
Replacements

[image]
[image]
[image]
[image]
[image]

VArcht, 2015-12-04:
We should do a complete image for the splash screen instead of ground/gcs/src/app/resources/tau_trans.png

tgreer, 2015-12-04:
Agreed.

VArcht, 2015-12-07:
more logos... https://github.com/d-ronin/dRonin/tree/next/artwork/Logo/logo.iconset

mlyle, 2015-12-17:
I don't see why multirotor-shapes needs branding / where it has any existing branding.

If that is good, let's close this issue (and handle other branding things as we encounter them).

closed 2015-12-18 by mlyle