d-ronin/dRonin #8: BrainFPV support

fujin, 2015-10-04:
With the relaxation of strict open-hardware policies it will be possible for us to have BrainFPV in-tree, with unified GCS, packaging, release timelines, if @BrainFPV is up for it.

What needs to be done to make this happen? Is everyone OK with the idea? Brain is a great, commodity board with excellent performance.

tgreer, 2015-10-04:
+1 to this. I love the brain hardware, it bugs me that (though no fault of @BrainFPV) there's a delay in stuff making it down to the brain tree.

jihlein, 2015-10-04:
Could possibly do the same with the Dodo, cleaning up (primarily removing the bootloader) the code I already have.

fujin, 2015-10-04:
@jihlein I have local modifications for that I had some success testing on spracing3, tornadof3 as well. I think supporting the f3-clone-derivative boards is also a good idea. Will log an issue. :+1:

fujin, 2015-11-25:
Latest @ #36

tgreer, 2015-11-26:
Closing this as #36 has now been merged. Yay!

closed 2015-11-26 by tgreer