d-ronin/dRonin #9: SPRacingF3, Dodo, TornadoF3: Modern acro board support

fujin, 2015-10-04:
All mentioned boards essentially run the spfr3 cleanflight. @jihlein has a branch in progress with naze32pro/dodo support. I spent a small amount of time adapting it to SPRacingf3. I also have some Tornadof3 here now which I believe runs the spfr3 as well, so potentially minimal modifications required.

The tornadof3 is my favorite of the bunch, np cp2102, signal llc buffer (unfortunately no ESC programming pass-thru), polulu 5v stepdown pin-out. It doesn't have sensors on SPI but I think the spi-as-usart stuff @mlyle has been talking about could be favorable. Especially if we can essentially settle on SPI-everywhere and then have usart-sensors targets speak SPI. :ocean:

tracernz, 2015-10-04:
I'm personally not in favour of adding a whole lot of targets. The criteria for adding a new target should require it to offer something that existing boards do not. In this case the Sparky1 appears to be a superior design already. If we do add new targets it would ideally be in preference of/replace an existing target where possible.

fujin, 2015-10-04:
This is basically just one commodity target with a few permutations of sensors. I feel like the *f3 boards are going to be pretty popular / widely available.

fujin, 2015-10-05:
I was wrong, tornado f3 is a separate branch / hardware target. https://github.com/MotoLab/cleanflight/commit/1a9489f3af1330ed4d1812aa50005afd39c286ed apologies :)

tracernz, 2015-10-05:
If we have a maintainer (@jihlein?) this may be viable. I'm sure we'll discuss it more on IRC before we make a decision. :D

jihlein, 2015-10-06:
I have no issues supporting it, my port to the Dodo does however need some further work before it could be called ready for prime time (need to remove the bootloader).  I guess the biggest question is will anyone use it, is it worth the effort?  I made the port for a neighbors son who liked what my Taulabs equipped Tricopter was capable of.  He wasn't satisfied with "Cleanflight" on his Dodo board, and I couldn't help him thru the setup.  Aiden is flying the Dodo with Taulabs pretty routinely and it seems to work well.  I get a real kick out of helping him and his younger brother.

The Dodo/Taulabs capabilities were mentioned on the Dodo forum on RCGroups, but there was no further interest from other users.  I don't know if that would change if it became an "official" target.

Thinking a little further, the Dodo port, minus the bootloader, gives back 16K of flash for other code features, and possibly frees up some heap space.  This brings a little more capability to an F3 board as I ran out of flash space on the Naze32Pro while adding OpenLog support.  I solved that by removing the support for NMEA GPS string decoding.......

jihlein, 2015-11-28:
I was toying with the idea of renaming my Dodo port to F3racing. With a few small modifications to the sensors and their initializations,  the port then could be used for the Dodo and both variants of the SPracing board.

tgreer, 2016-04-08:
@jihlein I'm still curious about this... I have another board http://shop.myairbot.com/index.php/flip32f3aio-lite.html which gets flashed with the SPRF3 Hex under cleanflight, and has a MinimOSD on board also. It's a neat all in one board, would be good to have this supported :)

jihlein, 2016-04-08:
@tgreer - I think the Dodo target in my "myNext" branch is very close.  

It would need an update to not use the baro, and possibly to the oscillator frequency.  Can you tell if the oscillator is 8 or 12 MHz?

I believe the rest of the IO is pin for pin compatible.  The only exception is pwm outputs 3 and 4 do not work, as they are the USB pins, and I have not completed removing the bootloader.  You can't use the the bootloader as the USB pins are not hooked up though.

tracernz, 2016-04-08:
If it wasn't for Naze flash size I'd almost be tempted to write a UART variant of the bootloader for these targets. :/

jihlein, 2016-04-09:
@tgreer - I disabled the BMP280 in the Dodo target.  I believe this should now run in the Flip32F3 board if it is truly SPRacingF3 compatible.  You can find this test branch at:

https://github.com/jihlein/dronin/tree/flip32f3aio

Build and load the ef_dodo.hex file.  Let me know if it works.  The only issue I can think of is the oscillator might be the wrong frequency.

The branch is based off the latest next.

nathantsoi, 2016-05-09:
awesome, has anyone tested this? cc @jihlein ?

jihlein, 2016-05-09:
@nathantsoi - I don't believe the Flip32F3 branch has been tested by anyone.  I do have a branch for the Dodo at: https://github.com/jihlein/dronin/tree/myNext.  The Dodo target was put together for a friend of mine.  An earlier version of it has been successfully flown.  I'm not sure the latest version with all the latest merges from the last release though, as I don't have a board to test it on.

FWIW, both of the above mentioned targets were derived form my Naze32Pro branch, which I still actively fly.

mlyle, 2016-06-12:
Closing-- we can evaluate targets on a case-by-case basis (just picked up RE1, picking up DTFc, have picked up Lux, so there's at least some progress in this area).

closed 2016-06-12 by mlyle