d-ronin/dRonin #12: There's code to integrate with the TL forum that should point at something of ours

mlyle, 2015-10-06:


fujin, 2015-10-06:
Autotune, notably

VArcht, 2015-10-06:
We could use my gist idea.

Have it post anonymous gists with a thumbprint that we can find back.

Then we can have a web page to search them by size, motor, whatever .

Can have a tau hall of fame, some peeps might like that... :)

tracernz, 2015-10-06:
That would be quite cool actually. Gist for crash dumps would also be good.

mlyle, 2015-10-06:
I never thought about gist for autotune.  That sounds nice.

VArcht, 2015-10-06:
Yeah. crash dumps is where I first thought to use them but guil was afraid people would post fake ones to mess with him.

bonus, could replace this stuff with something sane like json https://github.com/d-ronin/dronin/blob/next/ground/gcs/src/plugins/config/configautotunewidget.cpp#L124

mlyle, 2015-10-06:
While we're at it-- we should add pulldowns for vehicle size and how well it flew with the autotuned settings.. so we can search for good tunes for "200-300 size quads".

VArcht, 2015-10-06:
yeah, nice to have known enums

dustin, 2015-11-26:
#38 should hopefully be quick and easy.  We can own and query the data.  Maybe make a usable interface (I can make a non-usable interface to start).

tracernz, 2015-12-09:
Nothing pointing at Tau Labs anymore.

closed 2015-12-09 by tracernz