d-ronin/dRonin #18: Multiple profile support

fujin, 2015-10-07:
I have ships (quadcopters) that are both 3S, 4S, sometimes with a variety of props on a given day.

I'd like to be able to autotune the powertrains individually and store them in banks as well as have a way to switch between them. Ideally this would extend to battery monitoring and stuff too.

tracernz, 2015-10-08:
Yes.

jpahka, 2015-10-08:
+1 on profiles, although I'd like to be able to change manually also, e.g. depending on if gimbal is onboard or not

jhitesma, 2015-11-25:
My #1 wish is for profiles.  But definitely with manual change.  One my builds with profiles I like to setup one "tame" profile I can let noobs try, one with my preferred setup, and sometimes a third if I want to try something different but don't feel like lugging a laptop or a USB-OTG cable with me.

Personally I like stick commands for switching since I tend to bump switches accidentally, but I know some people <tracernz> would prefer switches.

tracernz, 2015-11-25:
Ideally we could specify conditions for each profile, which could include stick pos, stick command, LiPo cells etc. The first profile matched would be selected, or a default one if none match?

mlyle, 2015-11-25:
I think the first thing we need is just the ability to store multiple sets of configuration information.  That is, valid multi-instance object serialization to flash.  Then we can teach subsystems to use them.

mluessi, 2015-11-26:
Yes, what we need is multi-instance settings UAVOs. Relevant TL issue: https://github.com/TauLabs/TauLabs/issues/614

Malcurin, 2016-04-03:
I would love this on a acessory switch, Or even the auto detect muyself :)

lolorc, 2016-05-20:
+1 I'm always the one with special use cases, atm I use mwrate as a way to have two rate profiles : rate and mwrate. mwrate has been removed from next iirc