d-ronin/dRonin #38: Structured app for autotune results

dustin, 2015-11-25:
Instead of a long free-form forum thread with a bunch of text and hard to link to things that require login to even look at, let's create a simple web app (I like GAE) to which we can post autotune results, then index them by various fields, aggregate them, query, look for anomalies, alert on stuff, etc...

SystemsGuy, 2015-11-25:
Database would be cool - "show me all 250's with Cobra 2204/2300's" ;-)

tgreer, 2015-11-25:
This sounds good.

tracernz, 2015-11-25:
Are you happy to handle the app end of it @dustin?

dustin, 2015-11-25:
Yeah, I can make something simple to get started.

VArcht, 2015-12-02:
Where are we going to store these? Probably be some other uses for persistent data for other parts of the website too.

dustin, 2015-12-02:
I plan to make a gae app and use datastore.  Export is important as well.
Last time I did something like this, I had it feeding data out to a public
couchdb that anyone could subscribe to.

VArcht, 2015-12-02:
an attempt at the json https://gist.github.com/VArcht/25cc3de6f0d7cf695a58

tracernz, 2015-12-02:
I already had the JSON code mostly written as per IRC :-P. This is what I've got, combining a few of the fields from your format. I have removed the user entered aircraft description string since we already cover the aircraft characteristics in other fields.
```json
{
    "dataVersion": 1,
    "identification": {
        "pitch": {
            "bias": -0.041178399999999997,
            "gain": 9.0118799999999997,
            "noise": 139.03700000000001
        },
        "roll": {
            "bias": 0.0073014200000000003,
            "gain": 9.3093800000000009,
            "noise": 151.97
        },
        "tau": 0.028000000000000001
    },
    "tuning": {
        "computed": {
            "derivativeCutoff": 18,
            "gains": {
                "outer": {
                    "kp": 5.3632600000000004
                },
                "pitch": {
                    "kd": 6.4892699999999996e-05,
                    "ki": 0.0096096500000000008,
                    "kp": 0.0032906699999999999
                },
                "roll": {
                    "kd": 4.8193899999999999e-05,
                    "ki": 0.0071367999999999996,
                    "kp": 0.0024438799999999998
                }
            },
            "naturalFrequency": 5.7999999999999998
        },
        "parameters": {
            "damping": 1.1000000000000001,
            "noiseSensitivity": 0
        }
    },
    "uniqueId": "0x2f0037000157325237313620",
    "userObservations": "I didn't see anything",
    "vehicle": {
        "armLength": 0,
        "batteryCells": 4,
        "esc": "",
        "firmware": {
            "board": "Sparky",
            "commit": "1b66659e",
            "date": "2015-12-01T18:02:00Z",
            "tag": "mpl-autotunebuffering"
        },
        "motor": "",
        "type": "QuadX",
        "weight": 0
    }
}
```
I'm also adding a button to copy to the clipboard that will give you this (yet to add the new fields to it):
```
Flight controller:	Sparky
Firmware tag:		mpl-autotunebuffering
Firmware commit:	1b66659
Firmware date:		20151201 18:02 UTC

Aircraft description:
Wasn't me

Observations:
I didn't see anything

Measured properties:
		Gain		Bias		Noise
Roll:	9.30938		0.00730142	151.97
Pitch:	9.01188		-0.0411784	139.037
Tau:	28 ms

Tuning aggressiveness:
Damping:			1.1
Noise sensitivity:	1 %
Natural frequency:	5.8

Computed values:
			RateKp		RateKi		RateKd
Roll:		0.00244388	0.0071368	4.81939e-05
Pitch:		0.00329067	0.00960965	6.48927e-05
Outer Kp:	5.36326		-			-
Derivative cutoff:	18
```

dustin, 2015-12-02:
This is really great.  Thanks!

mlyle, 2015-12-02:
Looks good.. Maybe lose the 0x from uniqueid?

mlyle, 2015-12-02:
Also will "vehicle" be extensible?  e.g. we add more attributes later without code change?  I could see wanting to plumb like, output channel types, actuator curve fits, etc...

tracernz, 2015-12-02:
No reason you can't add more fields.

tracernz, 2015-12-02:
Okay, I've tweaked things slightly, renamed armLength to vehicle.size because armLength is multirotor specific and we support more than just multirotors. New clipboard format:
```
Flight controller:	Sparky
Firmware tag:		mpl-autotunebuffering
Firmware commit:	1b66659
Firmware date:		20151201 18:02 UTC

Aircraft description:
Type:			QuadX
Weight (AUW):	500 g
Size:			250
Battery:		4S
Motors:			Cobra 2204 1960kv
ESCs:			RG12

Observations:
I didn't see anything

Measured properties:
		Gain		Bias		Noise
Roll:	9.30938		0.00730142	151.97
Pitch:	9.01188		-0.0411784	139.037
Tau:	28 ms

Tuning aggressiveness:
Damping:			1.1
Noise sensitivity:	1 %
Natural frequency:	5.8

Computed values:
			RateKp		RateKi		RateKd
Roll:		0.00244388	0.0071368	4.81939e-05
Pitch:		0.00329067	0.00960965	6.48927e-05
Outer Kp:	5.36326		-			-
Derivative cutoff:	18
```
JSON format:
```json
{
    "dataVersion": 1,
    "identification": {
        "pitch": {
            "bias": -0.041178399999999997,
            "gain": 9.0118799999999997,
            "noise": 139.03700000000001
        },
        "roll": {
            "bias": 0.0073014200000000003,
            "gain": 9.3093800000000009,
            "noise": 151.97
        },
        "tau": 0.028000000000000001
    },
    "tuning": {
        "computed": {
            "derivativeCutoff": 18,
            "gains": {
                "outer": {
                    "kp": 5.3632600000000004
                },
                "pitch": {
                    "kd": 6.4892699999999996e-05,
                    "ki": 0.0096096500000000008,
                    "kp": 0.0032906699999999999
                },
                "roll": {
                    "kd": 4.8193899999999999e-05,
                    "ki": 0.0071367999999999996,
                    "kp": 0.0024438799999999998
                }
            },
            "naturalFrequency": 5.7999999999999998
        },
        "parameters": {
            "damping": 1.1000000000000001,
            "noiseSensitivity": 0
        }
    },
    "uniqueId": "2f0037000157325237313620",
    "userObservations": "I didn't see anything",
    "vehicle": {
        "batteryCells": 4,
        "esc": "",
        "firmware": {
            "board": "Sparky",
            "commit": "1b66659e",
            "date": "2015-12-01T18:02:00Z",
            "tag": "mpl-autotunebuffering"
        },
        "motor": "Cobra 2204 1960kv",
        "size": 250,
        "type": "QuadX",
        "weight": 500
    }
}
```
Screens:
[image]
[image]

Size could potentially be a dropdown that covers multirotor+heli sizes, thoughts? The tooltip currently explains that it's m2m for multirotors.

dustin, 2015-12-02:
I did the bare minimum for capture in https://github.com/d-ronin/autotown -- no display or rollup of any kind yet, but it can capture the results.  Once we get a few results in, we can start displaying them.  Also, any type of export people want should be possible.  Select some fields/constraints and csv, etc...

(well, it's slightly more than the bare minimum in that it uses queues to handle load or partial outages)

dustin, 2015-12-02:
Also, only a subset of the fields end up actually being decoded, but the raw captured data is stored for display/future proofing.  I stored the latest sample from above, for example:

[image]

mlyle, 2015-12-02:
This is exciting :)

VArcht, 2015-12-29:
Looking at using freeboard.io for presenting the results. Would have a list of results and a detail popup something like below.

[image]

VArcht, 2015-12-29:
dronin orange :)

[image]

mlyle, 2015-12-29:
It looks pretty.  I'd appreciate a little more contrast, even if it's bad for aesthetics-- this is something I'm going to be using a lot.

mlyle, 2016-01-06:
Closing as "good enough" -- the product-side dependencies are done, and the cloud services can update separately.

closed 2016-01-06 by mlyle