d-ronin/dRonin #41: UAVOs should be self-documenting

mlyle, 2015-11-25:
There's a lot of need to use the UAVO browser but it's not clear what fields mean without looking at the code.  It would be nice to have documentation strings in the XML that python & the GCS can use. (for tooltips etc).

tracernz, 2015-11-26:
Like so?
```XML
        <field name="TimerUpdateFreq" units="Hz" type="uint16" elements="6" defaultvalue="50">
            <description>How often your spinny things get a fresh command.</description>
        </field>
```

tracernz, 2015-11-26:
Okay I have this 90% done (the quote here was just to test that it is escaped properly).
[image]

tracernz, 2015-11-26:
The branch is at https://github.com/tracernz/dronin/tree/uavo-howtomakefly if anybody wants to check it out.

PR #60 (2015-11-29): UAVO: Add description element to fields, add tooltips in UAVO Browser
I was going to sleep but then I kept thinking about this :-P. Fixes #41.
UAVOs can now have an optional description element inside each field. This field is used in the GCS object browser to display tooltips on each UAVO field. No field descriptions are actually defined in this PR (an issue will be created once merged). To test you will need to add a description to one or more fields, e.g. in https://github.com/d-ronin/dronin/blob/next/shared/uavobjectdefinition/actuatorsettings.xml#L4:
```xml
        <field name="TimerUpdat [... cut]
commit dc4cd7e3 (2015-11-29): UAVO: Add description element for fields
And tooltips in GCS UAVOBrowser
Fixes #41
closed 2015-11-29 by mlyle