d-ronin/dRonin #42: Setup 'new contributor' labels based issue guidance

fujin, 2015-11-25:
See
[image])
 for an example from the InfluxDB project.

It would be great to have category and difficulty labels to provide a way for new contributors to find work they can pick up and do without worrying too much about a far-reaching, multi-target(?) change.

dustin, 2015-11-25:
I like this idea.  Finding low hanging fruit is good for both newbies and boredom.

fujin, 2015-11-25:
I started adding a bunch of labels.

tracernz, 2015-11-26:
@dustin difficulty/fix-cleanflight can be the hardest one ;-P

dustin, 2015-11-26:
Ha.  I try not to dislike stuff.  The organization is nice here, though.

fujin, 2015-11-27:
Closing this now :) we have a good bunch of labels and I see others are adding new ones.

closed 2015-11-27 by fujin