d-ronin/dRonin #43: Clarify who has what rights/role within the project

tgreer, 2015-11-25:
Currently I know 3 people have commit rights, but that's all I know about the structure and org of the project.

mluessi, 2015-11-26:
I think we should set up a simple / lean governance structure:

- An executive team of 3 members that can make decisions and delegate stuff
- The executive team gets elected by active members (developers, very active community members)
- Write some founding documents / rules. They can only be changed by a super majority of active members. The same process is used to call for re-election of the exec team.

This structure would be very simple and make sure that the exec team represents the active developers / members. Maybe we should do some work and look for inspiration from other open source projects.

fujin, 2015-11-26:
Something [like this](https://github.com/chef/chef/blob/master/CONTRIBUTING.md) for CONTRIBUTING.md could be useful as well, for roles not-necessarily-within-the-project (without the apache v2 specific stuff for now -- CLAs etc)

tgreer, 2015-12-01:
I will be putting some form of documentation together on this over the next few days.

tgreer, 2015-12-03:
I'm in agreement with @mluessi.

I propose:

* 3 person exec team, selected by the active members of the project.
* Serve for a 6 month period. At the end of the 6 months, the community can choose to keep the existing 3 or select a different 3 people to serve on the exec team.

Executive Team's Role:
* Guide the project forwards.
* Maintain the overall image of the Project.
* Maintain control over Project assets (including but not limited to: Github Org/Repo, Social Media).
* Controls access rights to Project assets, but not to reasonably withhold them from other members if requested.

fujin, 2015-12-03:
lgtm @tgreer

dustin, 2015-12-03:
lgtm

mlyle, 2015-12-03:
Sounds good-- it does something good enough for now and punts the problem 6mos-12mos down the road (I think we should plan on replacing this structure after one or two iterations with something a bit more thought out based on what real community membership we have an what problems we've experienced).

tracernz, 2015-12-03:
What's the definition of an active member and how do they select?

fujin, 2015-12-03:
@tracernz write / contributor / N recent PRs or docs updates / willing to show up to polls/video conferences I would expect at the least.

VArcht, 2015-12-03:
lgtm, at this point I think it is easy enough to agree on who is active.

tgreer, 2015-12-03:
@tracernz I'm going to use @mluessi's definition of that:
* active members (developers, very active community members)

How do they select? Members select the 3 people they deem best for the committee and the top 3 tallies get the spots?

tracernz, 2015-12-03:
Okay, so we should define a list of people who will be involved in the initial selection.

tgreer, 2015-12-03:
@tracernz indeed.

mlyle, 2015-12-03:
>  Members select the 3 people they deem best for the committee and the top 3 tallies get the spots?

Members select any people they're OK with being on the committee, top 3 tallies get the spots.  (They can choose 1 or 4.. i don't care, it doesn't increase their voting power).

tgreer, 2015-12-03:
Sorry, yes I should have put that in before, people have to say they're willing to stand before they can be voted for.

mlyle, 2015-12-04:
Eligible voters-- chath/droidicus/dsal/englishman/fujin/hshred/icee/jhitesma/juhis/laurent\/ohmeye/pappaduck/readerror/sfabris/simon--/systemsguy/tgreer/tracernz/trex_/varcht/vyizis 
              / ... pug398?  jihlein?

Any comment on initial voter list?  Apologies if I left anyone out.

tgreer, 2015-12-04:
As this has been agreed, and I won't be standing, I will be the returning officer for this. All those wishing to stand for election to the Executive Committee are to inform me via IRC or email they wish to stand by 1800GMT on Monday 7th December 2015.

tgreer, 2015-12-07:
Going to use opavote to run the vote online.

tgreer, 2015-12-07:
The following members have chosen to stand for election:

dsal
fujin
icee
tracernz
hshred

I will be doing the voting online, so I require an email to send the invite to.
Please PM me or email me with your email :)

tracernz, 2015-12-07:
@jihlein @pug398 ^

tgreer, 2015-12-07:
englishman does not want to vote.

tgreer, 2015-12-08:
Voting has now commenced.

tgreer, 2015-12-09:
Voting Complete.

Winners are hshred, icee, tracernz

See https://www.opavote.com/results/5689713507696640/0 for full results

dustin, 2015-12-09:
Woo!

mluessi, 2015-12-11:
Closing this as the exec board has been elected

closed 2015-12-11 by mluessi