d-ronin/dRonin #51: Make .uav exports diff friendly

dustin, 2015-11-25:
Minor changes often lead to very large diffs.  Need some kind of canonicalization pass.

mlyle, 2015-11-25:
I remember Kenn did some stuff in this area, to sort elements and attributes.

tracernz, 2015-11-27:
How about JSON?

dustin, 2015-11-27:
The format isn't all that important.  I just want small changes to make
small, easy to read diffs.

This does mean in any case some pretty printing and ordering is involved.

dustin, 2015-11-27:
It seems a fix has gone in.  Maybe this is no longer an issue.  I would occasionally get good diffs and occasionally rewrites.  I'll reopen this if it goes bad again.

closed 2015-11-27 by dustin

mlyle, 2015-11-27:
https://github.com/TauLabs/TauLabs/pull/1924 
https://github.com/TauLabs/TauLabs/pull/1833 

were the changes that improved this.