d-ronin/dRonin #58: dRonin Logo

VArcht, 2015-11-26:
dRonin needs a logo.

VArcht, 2015-11-26:
Concept 1:

[image]

jihlein, 2015-11-26:
I like it!!

mluessi, 2015-11-26:
I like the logo. However, I think it may be too complex. Ideally, the logo should look good using only 1 or 2 colors, so it can easily be printed on t-shirts, silkscreens etc. Also, it's good if the logo is compact / abstract, so it can e.g. be used in an app icon (without the d-ronin text).

fujin, 2015-11-27:
one color is also ideal for transparency on youtube (for example, the BrainFPV green.) I do like this though :)

mlyle, 2015-11-28:
I like the dR here.

http://www.dafont.com/digital-ninja.font?text=dRonin&psize=l

Maybe not the specific typeface, but.. the elements mirrored between the d and r, the diagonal line separating them, etc.

mlyle, 2015-11-28:
since this was not posted here: https://github.com/VArcht/varcht.github.io/blob/ol_test/images/logo_proof_01.png

VArcht, 2015-11-29:
Concept 2:

[image]

mlyle, 2015-11-30:
Flagging this here since it's liked:

[image]

fujin, 2015-12-03:
:+1: for :two: :five: logo25 from me!

VArcht, 2015-12-03:
Final

[image]

mlyle, 2015-12-03:
need consensus on final "ONIN" too

tracernz, 2015-12-03:
Perhaps a brighter/less brown orange would be good, and make it look less retro.

VArcht, 2015-12-03:
pick an orange?

mlyle, 2015-12-03:
i like current orange.  whatever we decide needs to work with white and black backgrounds

mlyle, 2015-12-03:
also after we pick a screen color we should pick a pantone color just for reference.

VArcht, 2015-12-03:
first goes at adding the onin.
[image]

[image]

mlyle, 2015-12-03:
Maybe something inbetween?  I like the weight of the second but there isn't enough gap between the characters to look OK on board silk, etc.

Also not sure i like the orange fill / black stroke on the line / perhaps that line has too much weight.

tgreer, 2015-12-03:
the bar above the o looks stupid.

VArcht, 2015-12-04:
Horizontal

[image]
[image]
[image]

VArcht, 2015-12-04:
Stacked

[image]
[image]
[image]

VArcht, 2015-12-04:
oh, and hat :)

[image]

mlyle, 2015-12-04:
I like https://cloud.githubusercontent.com/assets/828599/11591516/34fbccb6-9a4d-11e5-9ace-50d1f0696140.png

tgreer, 2015-12-04:
I don't like the bar over the O. other than that I agree.

VArcht, 2015-12-04:
I like H3 as well

tgreer, 2015-12-04:
The consensus appears to be H3.

mluessi, 2015-12-04:
I'm with @tgreer, I don't like the bar over the o. For me as a native German speaker, it looks like "dRönin". Also, I'm concerned that if the left part of the logo (dR with stuff around it) is printed in a single color with limited resolution (e.g. silkscreen) it may become one large blob and it may be difficult to make out the letters.

VArcht, 2015-12-04:
@mluessi 

[image]

mluessi, 2015-12-04:
@VArcht ok, that looks good :)

mlyle, 2015-12-05:
https://github.com/VArcht/varcht.github.io/blob/ol_test/images/dronin_logo_ob.svg

mlyle, 2015-12-10:
Closing in favor of other branding issue.  Logo is done, becoming familiar, and looks great!

closed 2015-12-10 by mlyle