d-ronin/dRonin #61: Add ability to only import non-default settings

mluessi, 2015-11-26:
Currently, when UAVO settings are exported and later imported, they overwrite all the settings. This leads to problems, e.g. the VTOL path follower defaults have been tweaked to work better, but users who import settings during an upgrade won't benefit from the new defaults.

I think it would be good to have the ability to only import settings that are different from the default. One way to do this is to indicate in the exported xml file which settings are different from the default. When importing, the user could then select to either import all settings or just the the ones that are different from default (the latter should be the default choice).

kubark42, 2015-12-23:
+1

It would be even cooler if default changes could be accompanied by a little text that helpsthe user understand if they potentially want to stick with the old default.

mlyle, 2017-02-06:
I think this shouldn't be done.  There are settings changes where moving a default in combination with user's settings that they did change that could be dangerous.

closed 2017-02-06 by mlyle

tracernz, 2017-02-06:
My thinking was a table with a 2-way checkbox for each thing that will/might need to change, options are "keep old value", "use new default".

mlyle, 2017-02-06:
I think I like our current practice-- changing the name of things that we should pick up new defaults for; otherwise not picking up new defaults for existing setups.

We could do something smart with a lot more UI work and explanation.. because it would be really unclear this time around, for example, whether people should pick up my derivGamma change or not.