d-ronin/dRonin #64: Social media presence / grab accounts

mlyle, 2015-11-27:
* [ ] Facebook (@jhitesma)
* [ ] twitter (@jhitesma)
* [ ] instagram (@dustin ?)

dustin, 2015-11-27:
dronin is taken on instagram, and no dashes.  So going with [flydronin](https://www.instagram.com/flydronin/) there (to match [twitter](https://twitter.com/flydronin)).

fujin, 2015-11-27:
I see https://www.facebook.com/flydRonin/ so checking off fb. @jhitesma

jhitesma, 2015-11-29:
Just confirming that I snagged flydronin on twitter and was able to re-use my phone number for it so API access read/write is possible.  (apparently twitter lets you do that on new accounts but not existing accounts - but I can only get SMS from one of the accounts that's using my number, no big deal.)  @mlyle also has the twitter login info so there's no single point of failure.

And on Facebook I snagged flydronin for the URL, the name on there is currently DRonin I put in a change request for "The DRonin Project" at @mlyle's suggestion but FB doesn't allow changing the name until something has been up for 7 days and they don't allow names with lower case first chars :-1: 

On FB @mlyle and I are listed as admins and @fujin and @SystemsGuy are listed as editors.

I set twitter to private for now, the FB page I haven't bothered since really neither have anything up yet.  I just used the image off the .org domain as placeholders until we have an official logo/identity.

jhitesma, 2015-11-30:
I looked into youtube and saw that "dRonin" is already taken.  However "flydRonin" is available.  But...to give more than one person access to a youtube channel it apparently has to be linked to a G+ page and pages are only for businesses (and apparently require an address now.)  I didn't see any pages for other FC projects on G+ - just a community for OP.  Maybe @dustin has more insight on this?  Or do we not want to worry about G+/YouTube.  I can see not worrying about G+ but YouTube seems like it could be useful...but only if we had multiple users.  We could just create a gmail account that manages the channel and share the login for it but that's ugly....

Just in case I went ahead and grabbed flydronin@gmail.com (and sent the password to @mlyle as well) since "dronin" was already taken.

tgreer, 2015-12-03:
Do we really need/want a youtube channel? Other than that, we've ticked the stuff we want so if nobody has a burning desire for a branded youtube channel we can close this issue.

dustin, 2015-12-03:
I think we should need a youtube channel before we worry about it.

tgreer, 2015-12-03:
Agreed. Closing.

closed 2015-12-03 by tgreer