d-ronin/dRonin #68: TX wizard: make robust to small movements/jitter

mlyle, 2015-11-28:


jhitesma, 2015-11-28:
This actually worked much better for me this time than the last time I tried doing it about 6-9 months ago.  However I did still experience some issues (Using 20150928 Brain release) with channels being mis-recognized when running the input wizard the first time. 

Thought that could be related to issue #69 since it worked after I changed the port from PPM to PPM+Outputs and re-ran the wizard

Video for reference: https://youtu.be/NTS6S9icEVI?t=4m21s  (same vid as in issue #69)

jhitesma, 2015-12-03:
And just flashed my Naze and am having a hell of a time setting up the inputs because of this.  Same basic RX I used on my Brain a few days ago (DIY FlySky PPM which is showing about 2-3 jitter on the signals)

wizard totally failed - detected things before I could touch anything.

And manual setup was impossible until I set the deadband to 3.0...even then it was jumpy despite the values only shifting 2-3.

jhitesma, 2015-12-16:
Just to update on this I determined that the RX I was using was having issues which were making the problems more pronounced.  

However even on my other RX's that aren't randomly cutting out but still have 2-3ms of jitter I still have issues with things like channels being "detected" in the setup wizard before I touch anything and with screens in my OSD switching even though I don't touch the switch I have assigned to do that.

PR #683 (2016-02-24): flight/receiver activity: double activity detection range
Make things less likely to grab the wrong channel as an interim measure
for Tanto release.

Fixes #68

[[image]](https://reviewable.io/reviews/d-ronin/dronin/683)
commit d51f12ee (2016-02-24): flight/receiver activity: double activity detection range

Make things less likely to grab the wrong channel as an interim measure
for Tanto release.

Fixes #68
closed 2016-02-24 by tracernz