d-ronin/dRonin #69: Vehicle setup wizard: Confusing for setups that require port configuration changes.

jhitesma, 2015-11-28:
When using the vehicle setup Wizard on a hex with a brain board I wasn't able to complete the output calibration since motors 5/6 are off the main port and that setting had not been changed yet.  I was able to set the values - but obviously the motors never spun up since the port wasn't set to enable those outputs yet.

Maybe a new step in the setup wizard to allow setting board options before doing calibrations?

mlyle, 2018-03-13:
We don't have many targets like this anymore.  I'm inclined to WONTFIX.  Any objections?

jhitesma, 2018-03-15:
I've swapped the hex to using a seppuku...still have the OG brain but undecided what it will go in...so...I'm at least not experiencing the issue anymore.  

Seems unlikely enough to bite others that it's probably not worth the time to hassle with.

closed 2018-03-15 by mlyle