d-ronin/dRonin #70: Usability: Input configuration

jhitesma, 2015-11-28:
Display of current TX values is disappointing for users coming from Base/Cleanflight.  In their configuration app the values are displayed as a bargraph that updates in real time making it very easy to tell how your TX/RX are working.  

In our GCS the numeric value is shown to the right - but it updates slowly and isn't obvious so it's easy for a new user to miss.  I've always found the delay in values updating to be really annoying when trouble shooting common issues like figuring out why stick arming isn't working.  

Going into manual calibration values display in real time.  But every time you go into manual calibration the existing setup is wiped and arming is disabled so you have to reset everything.  This feels really cumbersome and annoying when you're first coming over from naze-land and I still find it frustrating when I'm having an input calibration issue.  

Also min/max values can be input directly, but neutral values can only be set with a slider - which is hard to adjust in small increments.  It would be nice to be able to input neutral values directly without having to use the UAVO browser which is intimidating to new users.

DTFUHF, 2015-11-28:
This is a pretty old issue for me that never really got addressed, partially because the Qt way of doing this easily is ugly, and the Qt way of doing this nicely is hard.

tracernz, 2016-01-31:
Would be good to get this done for Tanto. I quite like the way Libre does this.

tracernz, 2016-07-23:
From #128:
> There's an awesome virtual transmitter that aids input channel setup greatly, but it's only available in the channel input wizard. Adding this to the input channel setup screen would help immensely in understanding which controller channel does what, if it's backwards, etc.

mlyle, 2017-01-13:
@tracernz did this!

closed 2017-01-13 by mlyle