d-ronin/dRonin #72: Usability: Basic setup instructions for new users

jhitesma, 2015-11-28:
I'm going to try and work on writing something up for this.  Basically a get in the air quick guide for new users and those coming over from other platforms.  Things I found (and still find) confusing coming over from the traditional Naze firmwares:

Input configuration:  (See Issue #70)
General GCS layout: Tabs vs. panes
Object Browser:  request/send vs. load/save
Changes that require a reboot vs. those that don't ( Think they may all have warnings now but need to check.)
Attitude Filter Settings:  (When to change and why - for beginners...leave them alone, but at least answer what they are.)

Most of this is in the wiki...but isn't obvious or easy to find.  So it's as much about re-organizing and writing new material.

jhitesma, 2015-11-28:
While I was opening this issue @mlyle commented in IRC:
12:18 AM <icee> so i'm a new user of android, coming from ios
12:18 AM <icee> i really like how when i first launch an app, i get a dumb tutorial screen highlighting where everything is
12:19 AM <icee> it would be nice when you first go into GCS, to put someone on the config tab, and be like.. "you're going to be changing stuff here (and will do config wizard, input wizard, and then _____) and you can look at the status of subsystems down here
12:19 AM <icee> just a little bit of orientation as to what's going on
12:19 AM <icee> a couple sentences plus a couple portions of the screen highlighted for what they are

mlyle, 2016-04-08:
Being handled in #897 doc issue for tanto.

closed 2016-04-08 by mlyle