d-ronin/dRonin #73: GCS: Change welcome screen panels

tracernz, 2015-11-28:
The current information shown on the welcome screen isn't particularly useful to end-users. We should consider replacing the Github panel with github issue titles and a link to the issue tracker. The TL forum stuff should be removed, possibly pending our own forum. If we keep the forum box it should be altered to display titles only so it fits more than one post. We should make it possible to display critical (nice bold red) announcements to users of older versions with known critical bugs.

tracernz, 2015-11-28:
Related to #12

tgreer, 2015-12-03:
I'm in agreement with removing the forum box, it's of no use in all honesty. We might not even have a forum.

dustin, 2015-12-04:
I really want to host some data through autotown.  I don't care as much how it gets there (but I can have automatic polling things if someone has ideas of what they want -- bugs filed, closed, etc...)

mlyle, 2016-03-04:
Done for now

closed 2016-03-04 by mlyle