d-ronin/dRonin #75: GCS: Analytics

tracernz, 2015-11-28:
Consider instrumenting GCS with (Google?) analytics. This would allow us to track usage of particular boards, features, and errors, allowing us to make more informed decisions and better data to combat frustrations.

dustin, 2015-12-17:
Just send a request for to autotown with some basic parameters and we're mostly there.

dustin, 2015-12-17:
If we can send off an HTTP POST on board connect, then it'd be really great to get the following:

* installation id (per @mlyle -- not sure if we have anything like this)
* platform
* gcs version info
* hardware type
* firmware version

This is probably enough stuff to give us an idea of what's out there.

dustin, 2015-12-17:
I'd like to squeeze in more info if we can do it in a way that people won't perceive as creepy.  It'd be very good for us to know what hardware is out there, what modules are in use, what flight modes are being used, etc...

In an ideal world where people didn't think we were targeting them individually, we could look for specific configuration issues that are known to be problematic (e.g. acro+ with gyro rate set to 1000).

tracernz, 2015-12-18:
It'd be nice to have an event log of buttons clicked etc. that can be correlated with crash reports.

mlyle, 2015-12-24:
The open PR has an event log that's suitable for knowing much more about usability.  We won't receive it in crash reports, though, right?  (Seems hard to make that actually happen).

PR #339 (2015-12-29): Usage statistics plugin

[[image]](https://reviewable.io/reviews/d-ronin/dronin/339)

edit by mlyle: fixes #75
closed 2015-12-29 by mlyle