d-ronin/dRonin #76: GCS: Notify when a reboot is pending (due to settings changes)

tracernz, 2015-11-28:
Add a notification and reboot button at the bottom of the config widget to advise when a reboot is pending due to config changes. This should be a big win for a common newbie frustration.

jpbarraca, 2016-02-18:
It would be very useful. This involves other widgets besides Board and Modules?

tracernz, 2016-02-18:
Off hand just the hardware settings and modules UAVOs, there are probably others too.

mlyle, 2017-03-20:
We could add a property on UAVO fields "requiresReboot", that when changed and stored a reboot is flagged required.  hardware settings, modules, and RGB LED count are definitely some.

tracernz, 2017-03-20:
Yes, that could work. Maybe a messagebox when you hit save asking if you want to reboot.. or just set an icon on the reboot button if we want less intrusive/annoying. I think I prefer the latter.

mlyle, 2017-03-20:
I was kinda thinking something down next to the telemetry widget, so it "follows you".

mlyle, 2018-03-18:
How about instead of uavo fields in UI, something that the uavogen understands?  Easy to add a tag <applyingRequiresReset/>

tracernz, 2018-03-18:
Yes, that is the right place to add "requires_reboot" information.