d-ronin/dRonin #77: GCS: Firmware upgrade wizard

tracernz, 2015-11-28:
The basic steps would be:
- Read all the settings UAVOs off the board
- Flash the new firmware
- Translate all the setttings UAVOs as much as possible (will need UAVO infrastructure changes)
- Send new settings to the board

The process should be fully automated so that the user doesn't need to click "halt", open a firmware file, click "flash", click "boot" etc.

tracernz, 2016-01-30:
The current wizard that is part of the setup wizard:
https://www.youtube.com/watch?v=y9icaF9MmBU&feature=youtu.be

mlyle, 2016-01-30:
We could use a web service for translating the uavos

tracernz, 2016-01-30:
Seems plausible with the git hash from the firmware version info. @dustin :-P

dustin, 2016-01-30:
I'd like to do that, sure.  What kind of interface are you looking for?

tracernz, 2016-01-30:
Send up an XML containing all the settings UAVOs. Web service fetches the UAVO definitions using the githash, does the same with the upgrade target. Compare field definitions and translates values between versions. Some possible considerations:
* Hash field definitions, copy values directly across if field def hasn't changed
* If field has changed a little (e.g. default value), copy and flag for user review
* For others set to defaults and flag review.

Send XML back with fields for straight copy, fields for review (maybe a little more info in this one to tell the user what exactly changed in the field def), and fields forced to defaults.

tracernz, 2016-01-30:
If the whole object hash is equal then not much point sending those ones back.

mlyle, 2016-01-30:
XML?  That would require the GCS to be able to read it.

Alternatively we could send up the binary UAVOs and IDs.

tracernz, 2016-01-30:
Yah, will need to send some metadata with the binary UAVOs though, may as well embed them in XML or JSON.

dustin, 2016-01-30:
Just give me some examples of what you want.

mlyle, 2016-04-06:
This is actually done!

closed 2016-04-06 by mlyle