d-ronin/dRonin #82: Quanton boot panic/loop

fujin, 2015-11-28:
User has reported Quanton boot failure / panic loop via IRC, investigate/fix.

```
09:01 <sfabris> don't know if it is expected, but my quanton is bootlooping with dronin :-(
09:02 <sfabris> (dronin next, just pulled)
```

sfabris, 2015-11-28:
GCS and firmware pulled from dronin:

Built on Nov 28 2015 at 19:56:55
From revision 87f49a9d 20151128
UAVO hash 86ca6bf0

From what fujin  is saying on irc should be a panic(1) (one brief blue led flash).

jihlein, 2015-11-28:
My Quanton seems to be working fine. Maybe eeprom needs to be cleared and the setup restarted from a clean state?

mlyle, 2015-11-28:
If it's one flash and not repeating it's not a panic.  But it could be an invalid-config related boot fault.

mlyle, 2015-12-03:
@sfabris -- any details?  does this still happen after config reset?

sfabris, 2015-12-03:
Can't try for a few days, as I am travelling. Will test and report once
back.

mlyle, 2015-12-15:
WORKSFORME  I suspect this is actually #83

closed 2015-12-15 by mlyle