d-ronin/dRonin #87: UAVOs: Add description to all settings fields

tracernz, 2015-11-29:
#60 added an optional description tag for each UAVO field. These are currently shown as tooltips in the GCS UAVOBrowser widget and are exposed in the Python API. Only a few example descriptions were added in the PR. The priority should be to complete settings objects first.

This issue is to write the remainder of the settings object field descriptions. Some example descriptions can be seen in e439f175b38fb57b68ca8d34bf08912cb884624b

Completed settings UAVOs (all fields):
- [x] ActuatorSettings
- [ ] AirspeedSettings
- [ ] AltitudeHoldSettings
- [ ] AttitudeSettings
- [ ] FaultSettings
- [ ] FixedWingAirspeeds
- [ ] FixedWingPathFollowerSettings
- [ ] FlightBatterySettings
- [ ] FlightStatsSettings
- [ ] GeoFenceSettings
- [ ] GroundPathFollowerSettings
- [ ] HomeLocation
- [ ] HoTTSettings
- [ ] HwAQ32
- [ ] HwBrain
- [ ] HwColibri
- [ ] HwCopterControl
- [ ] HwDiscoveryF4
- [ ] HwFlyingF3
- [ ] HwFlyingF4
- [ ] HwNaze
- [ ] HwQuanton
- [ ] HwRevoMini
- [ ] HwShared
- [ ] HwSparky
- [ ] HwSparky2
- [ ] HwTauLink
- [ ] I2CVMUserProgram
- [ ] INSSettings
- [ ] LoggingSettings
- [ ] ManualControlSettings
- [ ] MixerSettings
- [ ] ModuleSettings
- [ ] MWRateSettings
- [ ] OnScreenDisplayPageSettings
- [ ] OnScreenDisplayPageSettings2
- [ ] OnScreenDisplayPageSettings3
- [ ] OnScreenDisplayPageSettings4
- [ ] OnScreenDisplaySettings
- [ ] OpenLRS
- [ ] OpticalFlowSettings
- [ ] OveroSyncSettings
- [ ] PathPlannerSettings
- [ ] PicoCSettings (might be replaced so low prio)
- [ ] SensorSettings
- [ ] StabilizationSettings
- [ ] StateEstimation
- [ ] SystemIdent
- [ ] SystemSettings
- [ ] TrimAnglesSettings
- [ ] TxPIDSettings
- [ ] VibrationAnalysisSettings
- [ ] VtolPathFollowerSettings

raemin, 2016-01-26:
As part of the documentation effort, the wiki should be updated with the appropriate object names as quoted references are deprecated:

Spectrum binding : "DSM bind"
Autotune : "RelayTuningSettings"

tracernz, 2016-10-05:
I'm pretty sure this isn't actually fixed... Many of the fields are still not documented (although they all have at least empty description tags).

tracernz, 2016-10-05:
Updated list:
```
"Settings object field has empty description node! AirspeedSettings:ZeroPoint"
"Settings object field has empty description node! AltitudeHoldSettings:AttitudeComp"
"Settings object field has empty description node! AltitudeHoldSettings:Expo"
"Settings object field has empty description node! AttitudeSettings:AccelTau"
"Settings object field has empty description node! AttitudeSettings:VertPositionTau"
"Settings object field has empty description node! AttitudeSettings:YawBiasRate"
"Settings object field has empty description node! CameraStabSettings:FeedForward"
"Settings object field has empty description node! CameraStabSettings:MaxAxisLockRate"
"Settings object field has empty description node! CameraStabSettings:AttitudeFilter"
"Settings object field has empty description node! CameraStabSettings:InputFilter"
"Settings object field has empty description node! CameraStabSettings:FeedForwardTime"
"Settings object field has empty description node! CameraStabSettings:MaxAccel"
"Settings object field has empty description node! FixedWingPathFollowerSettings:UpdatePeriod"
"Settings object field has empty description node! GroundPathFollowerSettings:HorizontalVelMax"
"Settings object field has empty description node! GroundPathFollowerSettings:HorizontalPosPI"
"Settings object field has empty description node! GroundPathFollowerSettings:HorizontalVelPID"
"Settings object field has empty description node! GroundPathFollowerSettings:VelocityFeedforward"
"Settings object field has empty description node! GroundPathFollowerSettings:ManualOverride"
"Settings object field has empty description node! GroundPathFollowerSettings:UpdatePeriod"
"Settings object field has empty description node! HomeLocation:Be"
"Settings object field has no description node! HwAQ32:OSDPort"
"Settings object field has empty description node! HwAQ32:USB_HIDPort"
"Settings object field has empty description node! HwAQ32:DSMxMode"
"Settings object field has empty description node! HwBrain:USB_HIDPort"
"Settings object field has empty description node! HwBrain:DSMxMode"
"Settings object field has empty description node! HwBrain:GyroFullScale"
"Settings object field has empty description node! HwBrain:AccelFullScale"
"Settings object field has no description node! HwBrainRE1:RxPort"
"Settings object field has no description node! HwBrainRE1:SerialPort"
"Settings object field has no description node! HwBrainRE1:MultiPortMode"
"Settings object field has no description node! HwBrainRE1:MultiPortSerial"
"Settings object field has no description node! HwBrainRE1:MultiPortSerial2"
"Settings object field has no description node! HwBrainRE1:I2CExtBaro"
"Settings object field has no description node! HwBrainRE1:I2CExtMag"
"Settings object field has no description node! HwBrainRE1:ExtMagOrientation"
"Settings object field has no description node! HwBrainRE1:DSMxMode"
"Settings object field has no description node! HwBrainRE1:USB_HIDPort"
"Settings object field has no description node! HwBrainRE1:USB_VCPPort"
"Settings object field has no description node! HwBrainRE1:NumberOfLEDs"
"Settings object field has no description node! HwBrainRE1:CustomLEDColor"
"Settings object field has no description node! HwBrainRE1:LEDColor"
"Settings object field has no description node! HwBrainRE1:IRProtocol"
"Settings object field has no description node! HwBrainRE1:IRIDTrackmate"
"Settings object field has no description node! HwBrainRE1:IRIDILap"
"Settings object field has no description node! HwBrainRE1:FailsafeBuzzer"
"Settings object field has no description node! HwBrainRE1:HWRevision"
"Settings object field has no description node! HwBrainRE1:FlashID"
"Settings object field has empty description node! HwCopterControl:USB_HIDPort"
"Settings object field has empty description node! HwCopterControl:DSMxMode"
"Settings object field has empty description node! HwDiscoveryF4:MainPort"
"Settings object field has empty description node! HwDiscoveryF4:USB_HIDPort"
"Settings object field has empty description node! HwDtfc:USB_HIDPort"
"Settings object field has empty description node! HwDtfc:DSMxMode"
"Settings object field has empty description node! HwLux:Uart2"
"Settings object field has empty description node! HwLux:Uart3"
"Settings object field has empty description node! HwLux:USB_HIDPort"
"Settings object field has empty description node! HwLux:DSMxMode"
"Settings object field has empty description node! HwNaze:MainPort"
"Settings object field has empty description node! HwNaze:RcvrPort"
"Settings object field has empty description node! HwNaze:RcvrSerial"
"Settings object field has empty description node! HwNaze:DSMxMode"
"Settings object field has empty description node! HwNaze:MPU6050IntPin"
"Settings object field has empty description node! HwPlayUavOsd:USB_HIDPort"
"Settings object field has empty description node! HwQuanton:InPortSerial"
"Settings object field has empty description node! HwQuanton:USB_HIDPort"
"Settings object field has empty description node! HwQuanton:DSMxMode"
"Settings object field has empty description node! HwRevolution:RxPort"
"Settings object field has empty description node! HwRevolution:RxPortUsart"
[... 94 lines cut ...]
"Settings object field has empty description node! VtolPathFollowerSettings:HorizontalPosPI"
"Settings object field has empty description node! VtolPathFollowerSettings:HorizontalVelPID"
"Settings object field has empty description node! VtolPathFollowerSettings:EndpointDeadbandWidth"
[... cut ...]
"Settings object field has no description node! VTXSettings:Power"
"Settings object field has no description node! VTXSettings:Band"
"Settings object field has no description node! VTXSettings:Band_5G8_A_Frequency"
"Settings object field has no description node! VTXSettings:Band_5G8_B_Frequency"
"Settings object field has no description node! VTXSettings:Band_5G8_E_Frequency"
"Settings object field has no description node! VTXSettings:Airwave_Frequency"
"Settings object field has no description node! VTXSettings:Raceband_Frequency"
```

mlyle, 2018-03-13:
mlyle@midnight:~/dRonin/shared/uavobjectdefinition$ grep "description/" * | wc -l
460
mlyle@midnight:~/dRonin/shared/uavobjectdefinition$ grep "/description" * | wc -l
1074

70% done.. siiiigh.

PR #1400 (2016-10-05): UAVOs: Add descriptions
Credit to @yamahabest .  Supersedes #1335 

While this isn't perfect, super-speedy merge would be appreciated to reduce the opportunity for conflicts.  (This is already going to conflict with almost every existing PR.)

Fixes #87 
Fixes #1153