d-ronin/dRonin #97: discussion: remove flyingf4?

mlyle, 2015-11-30:
Cons-- there's some users out there, it's cheap
Pros-- it's difficult to set up (no sensors onboard unlike flyingf3), using user USB is problematic, Voh is problematic.

tracernz, 2015-11-30:
The user USB is designed for OTG host mode thus it can't power the board. There is a diode after the 3V3 regulator that drops Vdd to ~2.9 V which is well short of Vih on Atmel ESCs.

tgreer, 2015-11-30:
I'm all for removing targets that aren't widely used.

jhitesma, 2015-11-30:
I'm one of the people with one...and after what we learned about the DiscoverF4 board last night I'm in favor of removing it.  At least removing the info in the wiki about it if absolutely nothing else. 

Despite the STM docs saying it can be powered by USB it can only be powered off the ST-Link USB port not the "user" usb which is apparently designed for USB-OTG as tracernz explains above.

And while the STM docs say it can be powered off 5v directly I'm not really sure how since VCC is tied directly to the 3.3v net, and the 5v pins are for powering peripherals not putting power in.  I'm guessing through the ST-Link but haven't had time to really dig into it yet.

On the upside...apparently the F4 chips are more resilient than expected as mine has survived a year of flying with 5v being applied to the 3.3v bus!

Bottom line...despite having one I don't think we should be encouraging it and plan on dismantling mine.  Just not a good board for actually flying with.

jhitesma, 2015-11-30:
Oh, and I'm not sure I'd agree it's cheap.  It was for me because I already had all the sensors from a homemade multiwii.  But if you had to buy sensors, then $20 for the Discovery board and $10-$15 for a full set of sensors (baro/mag/accel/gyros) you may as well just grab a $35 sparky off aliexpress.  

Add in the other issues with it.  I'm totally in favor of dropping it even though I don't like the idea of dropping targets in general.

PR #101 (2015-12-03): flyingf4: remove target
not even compiled yet, just throwing this up here.

fixes #97
closed 2015-12-03 by dustin