d-ronin/dRonin #98: ChibiOS uptime is limited to 4294 seconds/1hr 11 min

tracernz, 2015-11-30:
ChibiOS:
https://github.com/d-ronin/dronin/blob/next/flight/PiOS/Common/pios_thread.c#L206
```c
#define ST2MS(n) (((((n) - 1UL) * 1000UL) / CH_FREQUENCY) + 1UL)
```
 Overflows at (2^32-1)/1000
 (2^32-1)/1000/1000(ms) = 4295 s = approx 1 hr 11 mins

FreeRTOS:
https://github.com/d-ronin/dronin/blob/next/flight/PiOS/Common/pios_thread.c#L43
```c
#define TICKS2MS(t) ((t) * (portTICK_RATE_MS))
```
portTICK_RATE_MS is 1 for our targets (i.e. 1000 Hz tick frequency) thus overflows at 2^31-1
 (2^32-1)/1000(ms) = 4294967 s = approx 1193 hours

tracernz, 2015-11-30:
We could fix this by going `#define portTICK_RATE_MS (1000UL / CH_FREQUENCY)` inside the Chibi block, moving TICKS2MS out of the FreeRTOS block and using it in the Chibi version of PIOS_Thread_Systime as well, i.e. doing the same on both OSs: https://github.com/d-ronin/dRonin/blob/ee240bf5f6c3211cc53f382847ce174ae0014bb0/flight/PiOS/STM32F4xx/Libraries/FreeRTOS/Source/portable/GCC/ARM_CM4F/portmacro.h#L113

tracernz, 2015-11-30:
Captured a log from the Quanton (ChibiOS), the last SystemStats captured before disconnect:
```
UAVO_SystemStats(name=UAVO_SystemStats, time=313.954, uavo_id=0x74E632D4, FlightTime=4294226, HeapRemaining=101340, EventSystemWarningID=0, ObjectManagerCallbackID=0, ObjectManagerQueueID=796797186, IRQStackRemaining=512, CPULoad=18, CPUTemp=0)
```
Confirms the theory IMO.

dustin, 2015-11-30:
That's way too close, yeah.  That thing FT showed recently gets like, four
hours, right?

mluessi, 2015-11-30:
Can confirm, have observed this on real HW.

tracernz, 2015-11-30:
Places where we definitely have problems:
https://github.com/d-ronin/dRonin/blame/next/flight/UAVObjects/uavobjectmanager.c#L1742
https://github.com/d-ronin/dRonin/blame/next/flight/Modules/ManualControl/transmitter_control.c#L1176 < The max value is wrong for Chibi, could potentially do some yucky things with arming if you're unlucky, and will result in a failsafe.

Stuff that won't work properly but doesn't look like it will break things too badly:
https://github.com/d-ronin/dRonin/blame/next/flight/Modules/OnScreenDisplay/onscreendisplay.c#L1663
https://github.com/d-ronin/dRonin/blame/next/flight/Modules/OnScreenDisplay/onscreendisplay.c#L1681
https://github.com/d-ronin/dRonin/blame/next/flight/Modules/UAVOMSPBridge/UAVOMSPBridge.c#L490
https://github.com/d-ronin/dRonin/blame/next/flight/Modules/VibrationAnalysis/vibrationanalysis.c#L280
https://github.com/d-ronin/dRonin/blame/next/flight/Modules/CameraStab/camerastab.c#L178
https://github.com/d-ronin/dRonin/blame/next/flight/Modules/GPS/ubx_cfg.c#L450 < I'm guessing this doesn't run later on..

Redundant code:
https://github.com/d-ronin/dRonin/blame/next/flight/Modules/Sensors/sensors.c#L222

Remaining to check:
```
Modules/OveroSync/overosync.c:178:	uint32_t lastUpdateTime = PIOS_Thread_Systime();
Modules/OveroSync/overosync.c:191:			if (!initialized && PIOS_Thread_Systime() < 5000) {
Modules/OveroSync/overosync.c:201:			updateTime = PIOS_Thread_Systime();
Modules/OveroSync/simulated/overosync.c:168:	uint32_t lastUpdateTime = PIOS_Thread_Systime();
Modules/OveroSync/simulated/overosync.c:187:			updateTime = PIOS_Thread_Systime();
Modules/OveroSync/simulated/overosync.c:216:	uint32_t tickTime = PIOS_Thread_Systime();
Modules/FirmwareIAP/firmwareiap.c:183:						lastResetSysTime = PIOS_Thread_Systime();
Modules/FirmwareIAP/firmwareiap.c:221:	return PIOS_Thread_Systime();
Modules/FirmwareIAP/firmwareiap.c:242:	if((uint32_t) (PIOS_Thread_Systime() - lastResetSysTime) > RESET_DELAY_MS) {
Modules/FirmwareIAP/firmwareiap.c:243:		lastResetSysTime = PIOS_Thread_Systime();
Modules/System/systemmod.c:564:	stats.FlightTime = PIOS_Thread_Systime();
Modules/System/systemmod.c:575:	uint32_t now = PIOS_Thread_Systime();
Modules/System/systemmod.c:577:		float dT = (PIOS_Thread_Systime() - lastTickCount) / 1000.0f;
Modules/System/systemmod.c:843:    uint32_t now = PIOS_Thread_Systime();
Modules/System/systemmod.c:844:    uint32_t timeToNextUpdate = PIOS_Thread_Systime() + MAX_UPDATE_PERIOD_MS;
Modules/System/systemmod.c:851:        	timeNow = PIOS_Thread_Systime();
Modules/Attitude/smallf1/attitude.c:190:			complimentary_filter_status = (PIOS_Thread_Systime() > 1000) ?
Modules/Attitude/smallf1/attitude.c:194:			(PIOS_Thread_Systime() < 7000) && 
Modules/Attitude/smallf1/attitude.c:195:			(PIOS_Thread_Systime() > 1000)) {
Modules/Attitude/smallf1/attitude.c:198:			accelKp = 0.1f + 0.1f * (PIOS_Thread_Systime() < 4000);
Modules/Attitude/smallf1/attitude.c:456:	uint32_t thisSysTime = PIOS_Thread_Systime();
Modules/Attitude/attitude.c:512:		attitudeSettings.AccelKp = 0.1f + 0.1f * (PIOS_Thread_Systime() < 4000);
Modules/Example/examplemodperiodic.c:86:	lastSysTime = PIOS_Thread_Systime();
Modules/Telemetry/telemetry.c:505:	timeNow = PIOS_Thread_Systime();
Modules/Telemetry/telemetry.c:628:			pausePeriodicUpdatesTime = PIOS_Thread_Systime();
Modules/Telemetry/telemetry.c:633:			pausePeriodicUpdatesTime = PIOS_Thread_Systime();
Modules/Telemetry/telemetry.c:668:	timeNow = PIOS_Thread_Systime();
Modules/Autotune/autotune.c:173:	uint32_t last_update_time = PIOS_Thread_Systime();
Modules/Autotune/autotune.c:217:				last_update_time = PIOS_Thread_Systime();
Modules/Autotune/autotune.c:233:				diff_time = PIOS_Thread_Systime() - last_update_time;
Modules/Autotune/autotune.c:238:					last_update_time = PIOS_Thread_Systime();
Modules/Autotune/autotune.c:250:				diff_time = PIOS_Thread_Systime() - last_update_time;
Modules/Autotune/autotune.c:284:					last_update_time = PIOS_Thread_Systime();
UAVTalk/uavtalk.c:626:        uint32_t time = PIOS_Thread_Systime();
UAVTalk/uavtalk.c:755:			uint32_t time = PIOS_Thread_Systime();
UAVTalk/uavtalk.c:1064:		uint32_t time = PIOS_Thread_Systime();
```

tracernz, 2015-12-01:
The killer:
```
PIOS_Recursive_Mutex_Lock (mtx=0x10000a8c, 
    timeout_ms=timeout_ms@entry=4294967295)
    at /home/mike/Dev/dronin/flight/PiOS/Common/pios_mutex.c:252
252		PIOS_Assert(mtx != NULL);
```
```
#0  PIOS_Recursive_Mutex_Lock (mtx=0x10000a8c, 
    timeout_ms=timeout_ms@entry=4294967295)
    at /home/mike/Dev/dronin/flight/PiOS/Common/pios_mutex.c:252
#1  0x0803e0c6 in UAVObjGetInstanceDataField (obj_handle=0x10000aa0, 
    instId=instId@entry=0, dataOut=dataOut@entry=0x200068bf, 
    offset=offset@entry=24, size=size@entry=1)
    at /home/mike/Dev/dronin/flight/UAVObjects/uavobjectmanager.c:1251
#2  0x0803e152 in UAVObjGetDataField (obj_handle=<optimized out>, 
    dataOut=dataOut@entry=0x200068bf, offset=offset@entry=24, 
    size=size@entry=1)
    at /home/mike/Dev/dronin/flight/UAVObjects/uavobjectmanager.c:1099
#3  0x0805653e in SystemAlarmsManualControlGet (
    NewManualControl=NewManualControl@entry=0x200068bf " \030\017")
    at /home/mike/Dev/dronin/build/uavobject-synthetics/flight/systemalarms.c:162
#4  0x08025f52 in set_manual_control_error (
    error_code=error_code@entry=SYSTEMALARMS_MANUALCONTROL_SETTINGS)
    at /home/mike/Dev/dronin/flight/Modules/ManualControl/transmitter_control.c:1245
#5  0x08026b32 in transmitter_control_update ()
    at /home/mike/Dev/dronin/flight/Modules/ManualControl/transmitter_control.c:289
#6  0x08025b8c in manualControlTask (parameters=<optimized out>)
---Type <return> to continue, or q <return> to quit---
    at /home/mike/Dev/dronin/flight/Modules/ManualControl/manualcontrol.c:130
#7  0x08021b9a in _port_thread_start ()
    at /home/mike/Dev/dronin/flight/PiOS/Common/Libraries/ChibiOS/os/ports/GCC/ARMCMx/chcore_v7m.c:255
```
@mlyle, is it possible the callback stack could overrun due to some weird clock rollover and wipe out the mutex (it's the next thing allocated after the cb stack)?

mlyle, 2015-12-01:
As we talked about on IRC, stack grows down on ARM.  So still a mystery.

tracernz, 2015-12-09:
This is a deeper issue than first thought. I'll have to investigate further. There's not really a good way to fast-forward to that uptime though because doing so breaks stuff in other ways, so it takes 1hr 11min every time you test.

tracernz, 2015-12-10:
Well I tried debugging this on the DiscoveryF4, but it doesn't suffer this issue despite being on F4! That narrows the issue down but means I need to hook the Quanton up to the debugger again.
```
(gdb) p vtlist.vt_systime
$9 = 7253364
```

tracernz, 2015-12-12:
This bug now just results in a broken telemetry connection but the board keeps running. Some kind of weird memory corruption seems to be happening, as if a stack is blown or something.

mlyle, 2015-12-15:
As far as we have been able to determine, this requires a telemetry connection up.  While not good, this makes this issue much less severe.

mlyle, 2016-01-10:
Deferring.

jpbarraca, 2016-02-18:
With a Sparky2 I noticed that the PicoC module will restart every 1h 11m. Will need to test it again.
My setup was:
- Had picoc with script that would blink the LEDs.
- Left Sparky2 powered without any input
- I noticed that sometimes the LEDs would flash
- Started registering the time between blinks and it matched 1h11m. Last time however it took around 15m.

Not sure that this is related. Just reporting.

mlyle, 2016-03-24:
Won't be further improving this before samsara; deferring.

mluessi, 2016-04-22:
Just did some tests: The crash also happens when telemetry is not connected. We should fix this for the next release; it's not impossible for a fix wing aircraft to fly more than 1h 11min.

pug398, 2016-04-22:
There are also some experimental "flying battery" multirotors that can also exceed 72 minutes flight time.

tracernz, 2016-04-22:
The cause is currently unknown.

tracernz, 2016-05-01:
On simulator, Attitude and Sensors watchdog flags timeout just after 4294 s.

closed 2016-05-07 by mlyle