d-ronin/dRonin #105: Buzzer Support

dustin, 2015-12-01:
I think we need a buzzer module with the following requirements:

* Configurable output (digital or pwm on something supported by the hardware)
* Configurable alarm conditions (beep on particular alarm conditions)
* User-initiated beeping (accessory is|isn't at a particular value)

Supporting naze's nfet thing would be nice.  [pwm buzzers](http://hobbyking.com/hobbyking/store/__14521__HobbyKing_174_Discovery_Buzzer.html) exist and should be quite easy to support.

See also:  https://github.com/TauLabs/TauLabs/issues/1349

tracernz, 2015-12-01:
Like https://github.com/d-ronin/dRonin/blob/next/flight/Modules/Actuator/actuator.c#L465

mluessi, 2015-12-01:
+1 Good idea. As @tracernz pointed out, there is already support for PWM buzzers. But it would be good to make this more general and configurable. It could also be removed from the "Outputs" section.

henn1001, 2016-01-31:
So should this be an module on its own? Or add it to the exististing actuator logic? Should be fairly easy to add with the new GPIO code that was added for picoC https://github.com/d-ronin/dRonin/blob/next/flight/targets/quanton/board-info/pios_board.h#L251

Just define some pins that can be used and list them in a similar way as the pwm Outputs.

mluessi, 2016-01-31:
@henn1001 I think this should be lightweight and not depend on PicoC. For example the naze32 (and derivatives) have buzzer pins and it would be nice if dRonin would support this feature.

mluessi, 2016-01-31:
Maybe it could be done similar to the LED config. I.e., a struct that defines which pin is used for the buzzer and a `PIOS_INCLUDE_BUZZER` that enables the feature on a specific target. By default, the buzzer should go on on receiver failsafe, so the resources used would be very minimal (so it would work on naze32 etc.).

henn1001, 2016-01-31:
Yeah - i meant without picoC. Those PIOS_GPIO defines can be used anywhere as far as i understood.

mlyle, 2016-01-31:
Just a question.. Would we work for non-PWM buzzers just with setting min and max to 0 and the period?

Trex4Git, 2016-01-31:
Nice to see some further interest in this.

When i started working on this (https://github.com/TauLabs/TauLabs/pull/1628) i had the impression that there was no really interest in this.
That  was  the reason i choose the PicoC way, and not a "core" module. 
I also activated GPIO Pins only on Qunaton, but  there was some discussion how it may be done on different targets.

henn1001, 2016-01-31:
@mlyle  More or less. https://github.com/d-ronin/dRonin/blob/next/flight/Modules/Actuator/actuator.c#L565 The code just toggles between min and max . So if 0 - 100% pwm value is selected it should just toggle between on and off / gpio behavior. But the functionality is a bit sparse - only lowBat and gps warnings. 

I´m doing the same in my picoC script and it works fine.

henn1001, 2016-02-21:
I´m still thinking about this. Is there enough room for a buzzer module on f1 boards?

tracernz, 2016-02-21:
Should fit on Naze, unsure about CC3D. It will be welcome for F3/F4 even if it doesn't fit on F1 :-).

dustin, 2016-02-21:
Does anyone use the existing buzzer stuff?  It's kind of in the way.  I
guess one workaround is to try to get, e.g. Naze buzzer to work within that
junk first.  It'd really be a lot nicer if all of that were removed, though.

jihlein, 2016-02-21:
Yes, I use it on my Naze32Pro port.  By luck, the pin with the transistor output on that board lines up with a PWM timer.  Use it for low battery warning and it works well.

But I agree, it could be replaced with something simpler.

dustin, 2016-02-21:
Wait, it does?  How do you configure that up?

jihlein, 2016-02-21:
After setting up the battery monitor, set the output like this:

[image]

It uses one of the PWM output pins.  Setup as shown above, the pin has a 100% duty cycle.  Set a lower duty cycle if it's too loud.  In most case you will need an external transistor to drive the buzzer, but in the case of the Naze32Pro, it's already on the board.  Unfortunately the Naze32 buzzer pin is not a PWM capable pin.  You could use a spare PWM output pin and use an external transistor though.

henn1001, 2016-03-01:
I have some barebone code  up and running and working
https://github.com/henn1001/dRonin/blob/BuzzerModule/flight/Modules/Buzzer/buzzer.c#L126
Now i just need to include some logic depended on some alarms :)

It´s actually really easy to to this - gpio pins can be pre defined the same way as this:
https://github.com/henn1001/dRonin/blob/BuzzerModule/flight/targets/quanton/board-info/pios_board.h#L251
https://github.com/henn1001/dRonin/blob/BuzzerModule/flight/targets/kkng/fw/pios_board.c#L595

And then turn them on/off with PIOS_GPIO_Off/ON

mlyle, 2016-09-05:
Targets with buzzer:  

* DTFc (low side bipolar, plain GPIO)
* Quanton (low side bipolar, DAC)
* Lux (UNKNOWN)

From CleanFlight:
```#define BEEP_GPIO   GPIOB
#define BEEP_PIN    Pin_13
#define BEEPER_INVERTED```

* Naze32 (low side bipolar, plain GPIO)
* Sparky2 (low side FETs, TIMER CHANNELs)
* AQ32 (low side bipolar "LED" channels)
* BrainRE1 (fpga managed)

mlyle, 2016-09-16:
I see the pr-open tag.  Unfortunately I intend to supersede that PR

henn1001, 2016-09-17:
Feel free to pick up where i left off : )