d-ronin/dRonin #121: GCS: Crash on closing

tracernz, 2015-12-03:
I can reproduce by:
- having no boards connected to PC,
- running GCS,
- switch to Flight Data tab,
- switch to Welcome tab,
- close GCS

```
0	QWindow::parent() const			0x7ffff691e860	
1	QWindowContainer::parentWasRaised(QWidget *)			0x7ffff71f16f0	
2	??			0x7ffff71f162e	
3	??			0x7ffff71f162e	
4	??			0x7ffff71f162e	
5	??			0x7ffff71f162e	
6	??			0x7ffff71f162e	
7	??			0x7ffff71f162e	
8	??			0x7ffff71f162e	
9	??			0x7ffff71f162e	
10	??			0x7ffff71f162e	
11	QWidget::raise()			0x7ffff71c8498	
12	QStackedLayout::setCurrentIndex(int)			0x7ffff71b7f65	
13	QStackedLayout::takeAt(int)			0x7ffff71b8924	
14	QLayout::removeWidget(QWidget *)			0x7ffff71b3aef	
15	Core::ModeManager::aboutToRemoveObject	modemanager.cpp	217	0x7fffc2592ca3	
16	Core::ModeManager::qt_static_metacall	moc_modemanager.cpp	136	0x7fffc25ddb92	
17	QMetaObject::activate(QObject *, int, int, void * *)			0x7ffff604c36e	
18	ExtensionSystem::PluginManager::aboutToRemoveObject	moc_pluginmanager.cpp	183	0x7ffff78f5dc5	
19	ExtensionSystem::Internal::PluginManagerPrivate::removeObject	pluginmanager.cpp	569	0x7ffff78e7943	
...	<More>				
```
Everything above 15 needs Qt library debug symbols which I don't have (yet). The object being removed is the "Welcome" tab (index = 0), and something downstream of the `m_modeStack->removeTab(index);` causes the crash. This may be deep in QML.

tracernz, 2015-12-03:
Removing the PFD from the Flight Data tab stops the crash so this is almost certainly a case of something QML related being released twice.

PTDreamer, 2015-12-03:
https://github.com/TauLabs/TauLabs/pull/2101

jhitesma, 2015-12-03:
Just tried to repo on windows build of cd1fb60 from yesterday and didn't see a crash.

PR #134 (2015-12-04): GCS: Fix racing condition which leads to a segfault on exit
See TauLabs/TauLabs#2101. Really nice work by @PTDreamer.

Fixes #121 

I confirmed that it does fix the scenario reported in #121.
closed 2015-12-04 by mlyle