d-ronin/dRonin #123: Usability: Wiki page with tips for users converting from [base/clean/beta]flight

jhitesma, 2015-12-03:
When I first tried Tau after using the various *flight's there were a number of things that tripped me up and weren't well documented.

The difference in how "rates" are handled was a big one.  The scaling *flight uses vs. the somewhat hidden settings in GCS (hidden in that they're not visible by default on the "basic" tab and new users may not want to look at a tab labeled "advanced" right off but that is a setting many will want to change.)

Difference in motor mappings is another.  I just flashed one of our builds and remembered that the default motor config is different.  To work with motors wired to my naze board as *flight expects I had to set:
1 -> Channel4
2 -> Channel2
3 -> Channel1
4 -> Channel3

The failsafe is also somewhat different and was confusing to me.  On *flight you set a timeout period for failsafe to kick in (which isn't configurable in ours unless it's buried in a UAVO, I haven't really dug) and a value for motor outputs that it will hold at, then a time value for how long it will hold at that before disarming.  I usually setup my *flight builds to failsafe after 1 second and to disarm after 1 more second.  On ours the best I can do is set the arming timeout to 10 seconds.  (this is probably another issue, I had raised it with Tau but was shrugged off...seems to me it would be nice to have a way to disarm faster than our timeout when the disarm is triggered by failsafe.  But I like the "fall like a rock" approach to failsafe.)

I'll add more thoughts as I remember them.

tracernz, 2015-12-03:
>Difference in motor mappings is another. I just flashed one of our builds and remembered that the default motor config is different. To work with motors wired to my naze board as *flight expects I had to set:
1 -> Channel4
2 -> Channel2
3 -> Channel1
4 -> Channel3

#124 improves this a lot.

jhitesma, 2015-12-03:
Yeah, but even with more understandable names it would be nice to have an easy reference for people switching from *flight to know what to plug in without having to think about it.  

Probably not as big of a deal for people who use servo connectors for their motors...but mine were just soldered right to the board so it was easier to remap than to rewire.

tracernz, 2016-04-05:
https://github.com/d-ronin/dRonin/wiki/User-Guide:-Migrating-From-*Flight

mlyle, 2016-04-08:
Remainder being handled in #897 tanto doc item

closed 2016-04-08 by mlyle