d-ronin/dRonin #128: GCS: Add visuals to input channel screen

DTFUHF, 2015-12-03:
There's an awesome virtual transmitter that aids input channel setup greatly, but it's only available in the channel input wizard. Adding this to the input channel setup screen would help immensely in understanding which controller channel does what, if it's backwards, etc.

mlyle, 2016-07-22:
Dup / closely related to #70

closed 2016-07-22 by mlyle

tracernz, 2016-07-22:
They are different issues really. #70 is to add an indicator on the channel sliders on the input form while this is to show the actual stick movement like in the Tx setup wizard. I think this one is a lot more useful for users.

mlyle, 2016-07-22:
They would share a lot of infrastructure (turning up update rate) and if we're going to do both, they should be accomplished at the same time to avoid laying it out twice-- so IMO one issue makes sense.

tracernz, 2016-07-23:
Okay, just need to cover the ideas in this one in the other one. 👍