d-ronin/dRonin #133: GCS:Stab Tab: Tweaks to basic page

tracernz, 2015-12-04:
We should add derivative to the basic page and give it a suitable scaling. This is a good page for ex-MultiWii users with this change. 

Should also review the limits as most people report that autotuned integral gains show up red.

mlyle, 2015-12-04:
Maybe boost the limits some.  Boring craft they're still normal-- it's just when you have crazy-high bandwidth that the integral is so big.

jhitesma, 2015-12-05:
I was going to suggest adding stick scaling/expo but that would add a lot more to the page leaving it not very different from advanced other than the scaling/interface.  So I'm tempted to suggest just ditching basic all together.  

Reason I suggest it is it's something that confused me coming from *flight and other people I've got to try Tau over *flight have been confused by it.  Increasing "rates" is one of the first things they tend to want to do and hiding it on the advanced tab just seems like we're burying important functionality.

The whole Basic/Advanced/Expert separation just feels kind of arbitrary to me.  Maybe it's just me though.

tracernz, 2015-12-05:
The scaling is the important thing. People don't deal well with a gain of 0.00000543.

PR #201 (2015-12-10): GCS: Tweak basic tab
Added derivative (with nice scaling), added max. rate and expo. Moved zero integral at zero throttle to advanced tab. Also increased the maximum settable rate from 500 to 1000 deg/sec.

(screenshot is pre-increased max rates)
[image]

Fixes #133

[[image]](https://reviewable.io/reviews/d-ronin/dronin/201)
closed 2015-12-10 by mlyle