d-ronin/dRonin #136: replace modelview + glclib with qt3d / qml

mlyle, 2015-12-04:
Should be a lot shorter, should kill a lot of our remaining opengl system compatibility/system requirements, and improve GCS performance.

tracernz, 2015-12-04:
I looked at this a little while ago. glc_lib generates a LOT of warnings with cppcheck! The only problem is that Qt3D isn't stable yet.

tracernz, 2015-12-05:
Also, I had the hack the glc_lib code to get it to load 3dxml files exported from solidworks. Qt3D looks like it supports a much wider range of file formats. I'd like to look at making some newer/more modern/more appropriate models for modelview some day.

tracernz, 2017-02-06:
Have a basic impl. working in the QMLViewer plugin.
Current issues:
- [ ] No way to get bounding box or otherwise scale the camera/view to model size. Plan is to scale each of the models to a uniform size.

tracernz, 2017-06-08:
Works okay inside a QTabWidget + QSplitter in a minimal demo app, but breaks GCS when inside a UAVGadget. Will need to look at UAVGadget stuff to see if it's doing something wrong/deprecated.