d-ronin/dRonin #143: Record Symbols for Breakpad

dustin, 2015-12-04:
In order to use breakpad, we have to have symbols exported from the build and available for the minidump processor.  As we're having crashes go to autotown, it makes sense to send the symbol tables up as well.  Perhaps this can also lead to analysis automatically occurring there.

dustin, 2015-12-04:
Putting this comment here since I don't know where else to put it.  :/

https://chromium.googlesource.com/breakpad/breakpad/+/master/docs/getting_started_with_breakpad.md

tracernz, 2015-12-04:
Yep, if we can keep this nice and simple and get it hooked into Jenkins it would be grand.

dustin, 2015-12-04:
I don't know what the thing looks like, but if it's just a file, I can make a tool that can be placed on builders along with a file for auth that hands it over.

dustin, 2015-12-17:
I hope this is easy.  I have a feeling we might regret having ~1/3 of the crash infrastructure, but having the symbols be useless.

tracernz, 2015-12-18:
We keep the git hash so I think that should allow us to build symbols for each dump.

mlyle, 2015-12-18:
We need to do it at original build time.  We have no guarantee for
byte-for-byte, address-for-address reproducibility just by having the same
code.

tracernz, 2015-12-19:
Yeah, but it will give us a shot at using our existing crash-dump data until we get something together to dump symbols from releases. ;-)

mlyle, 2015-12-19:
@tracernz have you seen this work?  Because I doubt it will work.

tracernz, 2015-12-19:
Nope, haven't tried. Will look into getting this sorted in the coming week if nobody else is?

mlyle, 2015-12-19:
If you have time, a recipe I could use as part of the package build to get things would be nice.

mlyle, 2016-01-06:
Needs appropriate documentation as part of build process and/or hooking it into the build.

mlyle, 2016-03-24:
Won't be further improving this before samsara; deferring.

mlyle, 2016-07-22:
This was done for OSX, but Windows still needs to have its path forward determined.

mlyle, 2016-07-22:
Closed in favor of #290 (more specific bug)

closed 2016-07-22 by mlyle