d-ronin/dRonin #148: GCS: Crash during flashing

tracernz, 2015-12-05:
Sequence:
* Halt, attempt to flash firmware, fails at first program step (after erase, which fails silently due to #147).
* Reboot board, wait for telem connection halt again.
* Attempt to flash board, GCS either hangs or segfaults while closing hid connection.
I'll capture backtrace next time I can get it to segfault.

tracernz, 2015-12-05:
```

Thread 37 (Thread 0x7fff82991700 (LWP 26605)):
#0  0x00007ffff536c12d in poll () at ../sysdeps/unix/syscall-template.S:81
No locals.
#1  0x00007fffb9e69248 in ?? () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
No symbol table info available.
#2  0x00007ffff5b66182 in start_thread (arg=0x7fff82991700) at pthread_create.c:312
        __res = <optimized out>
        pd = 0x7fff82991700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140735384459008, -8737840743461937015, 1, 132600112, 140735384459712, 140735384459008, 8738092759816347785, 8737862127477203081}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#3  0x00007ffff537947d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
No locals.

Thread 24 (Thread 0x7fff79960700 (LWP 26184)):
#0  0x00007ffff536c12d in poll () at ../sysdeps/unix/syscall-template.S:81
No locals.
#1  0x00007ffff39d1fe4 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#2  0x00007ffff39d20ec in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#3  0x00007ffff6062ae7 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#4  0x00007ffff600b252 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#5  0x00007ffff5e1a96c in QThread::exec() () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#6  0x00007ffff5e1f7df in ?? () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#7  0x00007ffff5b66182 in start_thread (arg=0x7fff79960700) at pthread_create.c:312
        __res = <optimized out>
        pd = 0x7fff79960700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140735233263360, -8737840743461937015, 1, 0, 140735233264064, 140735233263360, 8737553959927160969, 8737862127477203081}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#8  0x00007ffff537947d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
No locals.

Thread 23 (Thread 0x7fff7a161700 (LWP 26183)):
#0  0x00007ffff536c12d in poll () at ../sysdeps/unix/syscall-template.S:81
No locals.
#1  0x00007ffff39d1fe4 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#2  0x00007ffff39d20ec in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#3  0x00007ffff6062ae7 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#4  0x00007ffff600b252 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#5  0x00007ffff5e1a96c in QThread::exec() () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#6  0x00007ffff4b55d02 in ?? () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Quick.so.5
No symbol table info available.
#7  0x00007ffff5e1f7df in ?? () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#8  0x00007ffff5b66182 in start_thread (arg=0x7fff7a161700) at pthread_create.c:312
        __res = <optimized out>
        pd = 0x7fff7a161700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140735241656064, -8737840743461937015, 1, 0, 140735241656768, 140735241656064, 8737546293947408521, 8737862127477203081}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
[... 425 lines cut ...]
No symbol table info available.
#30 0x00007ffff6062ae7 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#31 0x00007ffff600b252 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#32 0x00007ffff6012d3d in QCoreApplication::exec() () from /home/mike/Dev/dronin/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#33 0x00000000004124d9 in main (argc=1, argv=0x7fffffffe3e8) at /home/mike/Dev/dronin/ground/gcs/src/app/main.cpp:435
        app = {<QApplication> = {<No data fields>}, static staticMetaObject = {d = {superdata = 0x7ffff7878380 <QApplication::staticMetaObject>, stringdata = 0x4316a0 <qt_meta_stringdata_SharedTools__QtSingleApplication>, data = 0x431800 <qt_meta_data_SharedTools__QtSingleApplication>, static_metacall = 0x415f82 <SharedTools::Qt [... cut ...] 5, cyan = 3341, magenta = 32125, yellow = 26214, black = 0}, ahsl = {alpha = 65535, hue = 3341, saturation = 32125, lightness = 26214, pad = 0}, array = {65535, 3341, 32125, 26214, 0}}}, time = {mds = 70598600}, message_number = 237, settings = <incomplete type>}
        plugins = {<QListSpecialMethods<ExtensionSystem::PluginSpec*>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = -1}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0x6f6800}, d = 0x6f6800}}

```

mlyle, 2016-03-05:
I wonder if this is still applicable.

tracernz, 2016-03-05:
Probably not due to the stuff you did. I suspect it was the same bug I was hitting with your PR that you fixed. Fixing #147 will make double-sure the situation doesn't occur too, and I think we should do this while we're bumping F3/F4 bootloader version.

mlyle, 2016-03-05:
Should we just close?

tracernz, 2016-03-05:
Yep.

closed 2016-03-05 by tracernz