d-ronin/dRonin #149: Autotune forum share button placement maybe not ideal

mlyle, 2015-12-05:
[image]

The lower right button is "close".

Usually the lower right button is what you want to do by default, "next" or whatever.  So this is counterintuitive.

dustin, 2015-12-05:
Yeah, the buttons aren't obvious.  It's also not obvious that it's safe to
close when you're done.

I want "just do both things and close" as the bottom right button.

mlyle, 2016-02-03:
Someone on IRC (I think @dustin) suggested a checkbox that says share, that defaults checked... that does so when you click close.

dustin, 2016-02-03:
I didn't suggest it, but I agree with it.  I think we're missing a lot of
shares there just because it's not even a thing that occurs to people is
possible.  And/or it's something they think they have to get perfect first.

This is also an area where a UUID for the installation would be useful, as
we could flag that and have messaging back to the install/board that says,
"Our mechanical turk has done further processing on your tune data and
noticed an opportunity for improvement..."

tracernz, 2016-02-03:
Usage stats should tell us if we're missing any..

dustin, 2016-02-10:
People don't seem to understand the share button.  I've seen some duplicates shares as well as untuned shares.  I get the feeling some people think it opts them in for sharing.

DTFUHF, 2016-02-10:
It'd be nice to have a link to the database as well!

dustin, 2016-02-10:
Yes, it (usually) returns the URL to the tune it just wrote. If we make
this not require a specific action from the user, that could get more
complicated.

PR #1044 (2016-06-24): autotune usability
Fixes #149

Todo:
* [x] Fixup 'done' vs 'finish' -- platform specific -- in final autotune dialog
* [x] Look at on Windows; consider pixmaps
* [x] Consider turning failure to converge red
* [x] Test many variations-- applying yaw, not, etc.
* [x] Test whole lifecycle on real hardware
* [x] Column width / maybe make dialog a little bigger
* [x] Disable button when not connected to autopilot

Done in this PR:

* detect new tunes, and invite user to apply them
* replace second autotune tab with pseudo-wizard
* simplify directions
* make it more likely users share
* do sharing a [... cut]
closed 2016-06-24 by tracernz