d-ronin/dRonin #156: Sparky2:Pathplanner:Setting a preprogrammed path (box) causes reset and setting not saved

pug398, 2015-12-05:
On every attempt Pathplanner can be enabled and put on switch but when 10M box is selected and saved fc resets and setting not saved.

pug398, 2015-12-19:
Still having this problem on Sparky2. Enable PathPlanner module. Put PathPlanner on switch. Reboot. Set system>settings>PathPlannerSettings>PreprogrammedPath>10mBox and save. Board resets and setting is lost. Can anybody repro on sparky2 or another fc?

mlyle, 2015-12-19:
I can't reproduce.

Note the way the UAVO browser works is really janky though.. You need to click on something else inside 'pathplannersetings' after selecting that value before clicking the disk to save.  Else it doesn't pay attention to the value you selected in the pulldown.

pug398, 2015-12-19:
Interesting. I just repro'd it on a new sparky1 just out of the bag. 
I will double check to see if it is something wonky I am doing...

PR #321 (2015-12-21): pathplanner: load settings asynchronously
Not yet tested.  Fixes #156 .

[[image]](https://reviewable.io/reviews/d-ronin/dronin/321)
closed 2015-12-21 by dustin