d-ronin/dRonin #157: Sparky2:RTH sequencer

pug398, 2015-12-05:
Perhaps a little more work can be done on the RTH sequencer. Specifically the pause and climb if below minimum altitude. How about this. Engage RTH. Pause for 1 second. Climb at a sane velocity to minimum altitude if not there already there. Pause additional time (some felt 10 seconds too long) and then start return path home. Currently it immediately climbs a few meters, pauses, climbs additional difference in minimum altitude and immediately starts return path. I am afraid to a new user it would appear broken.

pug398, 2015-12-06:
After thinking it over I believe I have flown certain scenarios where horizontal velocity was too high to have a RTH trigger, brake and settle in 1 second before rising. Splitting the delay in half we could maybe do hold, wait 5 seconds, rise to minimum altitude, hold, wait 5 seconds, fly path home, wait 5 seconds, land. Right now we manually trigger RTH and know we should slow down and stop but in the future there may be different options. If a new user triggers RTH and it holds, waits5, climbs, waits5, flies home, waits5 and then lands they would feel like it was an evenly spaced sequence. I would prefer that over the immediate short climb, wait10, long climb, fly home, wait10 and land. I am not sure that was the intended sequence. If it is already at minimum RTH altitude or above it could skip over the second 5 second delay. 

Other than this, pathplanner and known issues everything else has worked smoothly and kudos to making big changes and having the preview work as well as it has.

mlyle, 2015-12-06:
:) Cool you're flying the preview and liking it.  

My concern with maintaining the current altitude without at least some immediate rise is that it depends way too much on a single altimeter point-- if we get a low initial reading and are already close to obstacles it could go badly.  It could be a smaller initial rise amount though if you think that's best.

pug398, 2015-12-06:
I could go for an immediate climb to avoid obstacles but if it starts I would prefer it go all the way instead of start/stop. To me that feels like a hiccup even though it is told to do it. It does generate a wobbly arc in some cases if there is sufficient horizontal velocity but it would be climbing. The mind likes a little time to process what it is expecting to happen hence I mentioned the delay before jumping into action but if even a half to 1 second delay is felt unsafe then jump right in. The 10 second pauses seem like forever...

mlyle, 2015-12-10:
For consideration for the release, if able-- very slow immediate climb to target RTH altitude.

pug398, 2015-12-10:
You should probably get rid of the first 10 second delay in the middle of the climb

mlyle, 2015-12-10:
Yes, that's what what I meant-- immediate but slow climb to RTH altitude as a compromise between the different behaviors.

pug398, 2015-12-10:
That would be great :)

PR #279 (2015-12-20): vtolpathfollower: immediately (slowly) climb to target alt
And no longer dwell at target alt (For now).

Intended to fix #157

[[image]](https://reviewable.io/reviews/d-ronin/dronin/279)
commit 12df4f44 (2015-12-20): vtolpathfollower: immediately (slowly) climb to target alt

And no longer dwell at target alt (For now).

Intended to fix #157
closed 2015-12-20 by mluessi