d-ronin/dRonin #158: Do something sensible if user configures MSP on a port with MSP module disabled

dustin, 2015-12-05:
I was setting up my naze for autotune again real quick and managed to leave MSP configured up on the main port with the module disabled.  Now I get to boot pin and reflash.

Desirable alternatives:

1. Automatically enable MSP if MSP is configured to be in use.
2. If MSP is disabled, then having MSP on a port should automatically make it telemetry if no other telemetry is enabled.

Latter is a little hacky.

/cc @tracernz

tracernz, 2015-12-05:
This one fixes your issue here:
https://github.com/tracernz/TauLabs/commit/d0fecdf480f3e67373b3889cdfd2ee065acde9fa
and this one detects 57600 UAVTalk packets while running 115200 MSP:
https://github.com/tracernz/TauLabs/commit/b301a50b0529438a9b8f2115701ec32a2f36139b

I should make a PR.

tracernz, 2015-12-05:
Oh, I see the latter one includes the first one as well.

PR #159 (2015-12-08): UAVOMSPBridge: Detect 57600 baud UAVTalk packets
Also, give the port back to UAVTalk if the MSP module is disabled so the board doesn't get stuck in a bricked state where it can't be connected to GCS. Mostly useful on the Naze32.

This is some hacks I did in my local branch a while back. Haven't tested it since cherry-picking into dronin tree. Fixes #158.

[[image]](https://reviewable.io/reviews/d-ronin/dronin/159)
closed 2015-12-08 by dustin