d-ronin/dRonin #160: fujotune in UI

dustin, 2015-12-05:
@fujin has this yaw tuning concept that he claims is awesome.   It might be nice to put something in the UI to apply it if people think it's good.

```lisp
(defun fujotune (rp pp ri pi)
  (message "yp: %f\nyi: %f"
           (+ (max rp pp) (abs (- rp pp)))
           (+ (max ri pi) (abs (- ri pi)))))
```

fujin, 2015-12-06:
lmfao :+1:

mluessi, 2015-12-07:
@dustin can you elaborate?

dustin, 2015-12-07:
We currently autotune roll and pitch, but leave yaw at defaults. @fujin
reports good results from this heuristic.  If it general provides better
results than defaults, and doesn't have downsides, we should make it easy
for the used to apply these values.

mlyle, 2015-12-07:
I don't think there's a good underlying justification for "take the higher plus the delta" gains.

"Take the higher gain plus 10%" might be justifiable, which is probably the same thing in practice (probably all of @fujin's frames have a similar difference between roll & pitch).

mluessi, 2015-12-07:
Can we just add yaw to autotune?

mlyle, 2015-12-07:
It's nontrivial and harder.

dustin, 2015-12-07:
Didn't peabody do work on this?  I haven't kept up, but something in my brain tells me this was attempted and the results weren't good.

fujin, 2015-12-08:
I don't think @peabody124  got it working right. :( I've just been 'fujotuning' everything for Yaw since. Point of order is that my frames typically have a massive bias toward pitch, or negative bias along roll - there is typically much less mass along the roll axis compared to both battery and hdcamera along pitch. I do suspect that most builds end up with the same distribution (which as @mlyle says, > probably all of my frames have similar differences.). But yes, this is often the same across different frames.

The underlying justification for me is that fujotune flies better than default yaw PIDs and there is not really a better 1-click/open bc/calculator alternative - classic "PID tuning" for Yaw is pretty hard to do, since Kp doesn't really work the same as pitch/roll IME. I'd welcome a more scientific approach, this would certainly save some time and would be trivial to calc in the GCS. Even if it's not perfect for everyone. Similar to 'lock roll & pitch' checkbox I think.

mlyle, 2016-04-10:
This is actually done already :)

closed 2016-04-10 by mlyle