d-ronin/dRonin #163: Safe Boot consistently crashes GCS for me

dustin, 2015-12-06:
```
Process:               dRonin GCS [54947]
Path:                  /Users/USER/*/dRonin GCS.app/Contents/MacOS/dRonin GCS
Identifier:            org.dronin.gcs
Version:               1.3.1 (1.3.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           dRonin GCS [54947]
User ID:               238135

Date/Time:             2015-11-29 20:22:34.294 -0800
OS Version:            Mac OS X 10.11.1 (15B42)
Report Version:        11
Anonymous UUID:        61497D56-BD3F-F614-1B44-4A3037CE86D2

Sleep/Wake UUID:       47B57FB4-C7D3-4082-8D10-08E45BFEB5BC

Time Awake Since Boot: 210000 seconds
Time Since Wake:       26000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_pthread.dylib       	0x00007fff9d20e48d pthread_mutex_lock + 0
1   com.apple.CoreFoundation      	0x00007fff9a5947f1 CFRunLoopStop + 49
2   libRawHID_debug.dylib         	0x000000011919a3e3 hid_device_removal_callback + 51 (hidapi_mac.c:559)
3   com.apple.framework.IOKit     	0x00007fff90fe3cbd __IOHIDDeviceNotification + 187
4   com.apple.framework.IOKit     	0x00007fff90fbfcb9 IODispatchCalloutFromCFMessage + 256
5   com.apple.CoreFoundation      	0x00007fff9a566242 __CFMachPortPerform + 290
6   com.apple.CoreFoundation      	0x00007fff9a566109 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__ + 41
7   com.apple.CoreFoundation      	0x00007fff9a566079 __CFRunLoopDoSource1 + 473
8   com.apple.CoreFoundation      	0x00007fff9a55daab __CFRunLoopRun + 2171
9   com.apple.CoreFoundation      	0x00007fff9a55cfc8 CFRunLoopRunSpecific + 296
10  com.apple.HIToolbox           	0x00007fff91360d55 RunCurrentEventLoopInMode + 235
11  com.apple.HIToolbox           	0x00007fff91360b8f ReceiveNextEventCommon + 432
12  com.apple.HIToolbox           	0x00007fff913609cf _BlockUntilNextEventMatchingListInModeWithFilter + 71
13  com.apple.AppKit              	0x00007fff97d28d96 _DPSNextEvent + 1067
14  com.apple.AppKit              	0x00007fff97d281c5 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
15  com.apple.AppKit              	0x00007fff97d1cd28 -[NSApplication run] + 682
16  libqcocoa.dylib               	0x00000001137327fd QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2189
17  org.qt-project.QtCore         	0x00000001102ab8cd QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
18  org.qt-project.QtCore         	0x00000001102ae78a QCoreApplication::exec() + 346
19  org.dronin.gcs                	0x000000010f24b4f7 main + 11111 (main.cpp:435)
20  libdyld.dylib                 	0x00007fff8f31a5ad start + 1

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff9fd670a2 kevent_qos + 10
1   libdispatch.dylib             	0x00007fff9c1fe11d _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff9c1fdd85 _dispatch_mgr_thread + 52

Thread 2:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff9fd60c96 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff9fd600d7 mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff9a55e2b4 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff9a55d77c __CFRunLoopRun + 1356
4   com.apple.CoreFoundation      	0x00007fff9a55cfc8 CFRunLoopRunSpecific + 296
5   com.apple.AppKit              	0x00007fff97e7ed49 _NSEventThread + 149
6   libsystem_pthread.dylib       	0x00007fff9d2109b1 _pthread_body + 131
7   libsystem_pthread.dylib       	0x00007fff9d21092e _pthread_start + 168
8   libsystem_pthread.dylib       	0x00007fff9d20e385 thread_start + 13

Thread 3:: com.apple.CFSocket.private
0   libsystem_kernel.dylib        	0x00007fff9fd66222 __select + 10
1   com.apple.CoreFoundation      	0x00007fff9a59b4aa __CFSocketManager + 762
[... 618 lines cut ...]
JS JIT generated code              388K       17 
JS VM register file               24.0M       10 
JS garbage collector              3456K       12 
Kernel Alloc Once                    8K        3 
MALLOC                           328.4M       86 
MALLOC guard page                   32K        7 
Memory Tag 242                      12K        2 
Memory Tag 251                      24K        3 
OpenCL                               8K        2 
OpenGL GLSL                        256K        4 
Process Corpse Info               2048K        2 
STACK GUARD                       56.1M       22 
Stack                             17.7M       24 
VM_ALLOCATE                        596K       30 
__DATA                            29.4M      379 
__GLSLBUILTINS                    2588K        2 
__IMAGE                            528K        2 
__LINKEDIT                       116.9M      101 
__TEXT                           245.7M      379 
__UNICODE                          552K        2 
mapped file                       55.1M       39 
shared memory                     16.4M       11 
===========                     =======  ======= 
TOTAL                            924.7M     1320 
 ```

dustin, 2016-03-05:
Just repeated this on my sparky.

dustin, 2016-03-05:
Reflashed to latest and tried again just to be safe.  Crashed.

mlyle, 2016-03-05:
Interesting.  Thanks for trying.

PR #790 (2016-03-13): ground: hidapi: attempt to fix synchronization
edit: according to @dustin fixes #163 

This change is [[image]](https://reviewable.io/reviews/d-ronin/dronin/790)
closed 2016-03-13 by tracernz