d-ronin/dRonin #164: Add Team Black Sheep BST telemetry support

mluessi, 2015-12-06:
The [TBS Crossfire LRS](http://team-blacksheep.com/products/prod:crossfire_tx) uses a new telemtery protocol based on I2C. It would be nice to add support for the protocol.

Reference: https://github.com/PX4/Firmware/pull/2939

tgreer, 2016-12-11:
I can test this if someone can code it.

tracernz, 2016-12-12:
It's a bastardised version of MSP IIRC?

jhitesma, 2016-12-12:
I seem to remember something about it not quite being I2C just based on I2C with some extra goofiness added in.

Siytek, 2017-08-07:
Did the BST implementation every get completed? Last time I checked perhaps 9 months ago and it was still only sending the minimal messages. Glancing over bst.cpp it looks to be the same.

Would be good to have this support to free up the USART on the xFire receiver for SBUS.

mlyle, 2018-04-20:
Is there any strong reason to do this now that we support Crossfire Telem?

glowtape, 2018-04-20:
Sends exactly the same things over BST, AFAIK. If the use case is solely Crossfire, instead of other stuff using BST (what is there, anyway? just their GPS?), it seems pointless to implement it.

mlyle, 2018-04-20:
I think you can change VTX channels over BST, but we prefer smartaudio to that.

Closin'.

closed 2018-04-20 by mlyle

tracernz, 2018-04-21:
BST was never a good protocol to begin with. We already have better alternatives. :+1: