d-ronin/dRonin #169: Pathway for ChibiOS Adoption

tracernz, 2015-12-06:
Several targets are still using FreeRTOS. We should come up with a plan to move these onto ChibiOS. Additionally we need to decide if F1 will move to Chibi. We have noted increased RAM usage in some tasks with Chibi (although overall RAM usage seems to be low at least on F4 targets). I think a good approach here would be to convert either Naze32 or CC3D and compare real changes in resource usage and then make a decision. If we do convert everything to Chibi we could then begin replacing quite a few of the ST peripheral library drivers with ChibiOS HAL drivers. I am happy to do the DiscoveryF4 conversion ASAP leaving just F1 targets.

Targets still on FreeRTOS:
* discoveryf4
* coptercontrol
* pipxtreme
* simulation
* naze32
* revomini (pending #166)

mlyle, 2015-12-06:
Not simulation, right?

tracernz, 2015-12-06:
I just did grep -r PIOS_INCLUDE_FREERTOS and those were the targets that showed up, didn't look further. ;-)

mlyle, 2016-09-24:
With discussion of fundamental changes to F1 support, it would be just pip that needs particular attention-- post-artifice.

PR #1649 (2017-03-25): Big internal refactor of sim / freertos related things
Open for transparency / "dibs":

* [x] Move sim to direct posix threading / scheduling primitives
* [x] Move pipxtreme from freertos to ChibiOS
* [x] Remove freertos
* [x] Begin cleanup of freertos-related code
* [x] ~~Try to identify places that should have the chibios epilogue (uart? usb?) to trigger resched~~ defer to #1650 
* [x] ~~Anomaly: Some tasks on pi seem to not be realtime-class?~~
* [x] Need to somehow ensure precedence of Pi command line args
* [x] Anomaly: make sure pipx still works good

Fixes #1230 
Fixes #169
closed 2017-03-25 by tracernz