d-ronin/dRonin #182: Meta Controls Level Up

fujin, 2015-12-07:
- [ ] ghettomixer (no clipping with 1.0 throttle)
- [ ] 3d mode (reversible ESCs)
- [ ] "AIR MODE" PIDs at low/configurable/zero throttle for hangtime stabilization.

fujin, 2015-12-07:
@mlyle you've mentioned all of these in the past so wanted to collect 'em somewhere. :)

dustin, 2015-12-07:
Problem with air-mode is I sometimes really don't want that, but most of the times I do.  It seems like it might make landing hard.

I flew my naze32 with dRonin and an old 180 with baseflight over the weekend.  I cut throttle on both.  dRonin started to dip a little.  baseflight flipped.  Might be comparative balances just working out.

I do like my hangtime, though.  And I sometimes can't lose altitude fast enough because I fear what happens when I cut throttle too much.  It'd be nice to get rid of that as long as we can still land well.

mluessi, 2015-12-09:
How does this "AIR MODE" stuff work? Any links / documentation?

DTFUHF, 2015-12-09:
As far as I know, it's Facebook's terminology for a minimum positive thrust state where the copter is still stabilized, so you can descend quickly ( as fast as possible while still oriented upwards ) while keeping orientation stability.

tracernz, 2015-12-10:
So the PID controller remains active while at minimum throttle. Cleanflight used to have this a while back but they took it away because people complained about the behaviour when landing (the integral causes the motors to wind-up and sort of jump each time the vehicle contacts the ground). It seems it's being re-invented again under a different name.

tracernz, 2015-12-10:
I propose we split this issue into three separate ones, this one can be renamed to "Preserve control at minimum throttle" or something equally meaningful, since the discussion has been mostly on that topic, #211 covers 3D, and we should create one for ghettomixer with a more descriptive title about what it should actually do.

mlyle, 2015-12-10:
@tracernz Yes, I think of it this way:

* Ghettomixer with negative clip management that ensures that it is only willing to add 3-5% power for stabilization.  so a 4% throttle can never become more than, say, half the motors at 0% and half the motors at 14%.  (This may be *less* stabilization than before in some circumstances, but more than before when at 0%.  So this should also reduce the tendency to climb at low power from control oscillation/overtuning)
* Significant integral attenuation at very low throttle

DTFUHF, 2015-12-10:
What's ghettomixer? I'll probably never use it if you keep that name. Is it going to steal my wallet?

mlyle, 2015-12-10:
Ghettomixer is a clipping management and actuator scaling system that is implemented "wrong" -- proof of concept --- but yields benefits in many circumstances.  I really mean ghettomixer's successor with correct clipping management.

tracernz, 2015-12-10:
Ok, make an issue for it. It's hard to track when their are multiple issues within one github issue and the title is meaningless. ;-)

fujin, 2015-12-10:
Logged the other two issues. Closing.

closed 2015-12-10 by fujin