d-ronin/dRonin #194: OSD Menu: PID values are limited

mluessi, 2015-12-08:
BrainFPV TL Bug: https://github.com/BrainFPV/TauLabs/issues/33

When the user uses PID values that are outside of the range supported by the OSD menu, the menu limits it to the range supported by the menu, which is not ideal. The menu should have a way to detect if the PID terms are outside of the range and increase the range automatically without changing the PID values.

tracernz, 2015-12-13:
Perhaps we should think about making the data from the limits attribute available in the generated flight code. https://github.com/d-ronin/dRonin/blob/next/shared/uavobjectdefinition/stabilizationsettings.xml#L14

tracernz, 2015-12-13:
Related to my post above: https://github.com/TauLabs/TauLabs/issues/888

PR #276 (2015-12-16): FIX: Remove PID limits in OSD menu
Tested on brain using the OSD menu with some ridiculous PID values.

Fixes #194

[[image]](https://reviewable.io/reviews/d-ronin/dronin/276)
closed 2015-12-16 by mlyle