d-ronin/dRonin #198: GCS/Flight: Remove temperature compensation infrastructure

mlyle, 2015-12-09:
Only works on F1, poorly tested, uses up a tab in GCS.

tracernz, 2015-12-09:
So this is like the first suggestion you will read on OpenPilot forums re: getting nav. working properly, is it all just rhetoric?

mlyle, 2015-12-09:
@tracernz I've never done it, and we seem to fly nav without.

Original coptercontrol needed this, I bet.

mluessi, 2015-12-09:
@mlyle why do you think this only works on F1? The code is definitely used on all targets.

mlyle, 2015-12-09:
@mluessi I was wrong.  It's factored funny-- it lives down in attitude.c on F1 but in sensors.c on F3.  This is further proof that F1 and F3/F4 targets need to be adjusted to have the common code factored common rather than duplicated into different places as currently happens.

closed 2015-12-09 by mlyle