d-ronin/dRonin #204: Rename Revomini to Revolution

mluessi, 2015-12-09:
The OP Revolution target had been named Revomini, which is confusing to users as this name is largely unknown. I suggest we rename it to Revolution or maybe OP-Revolution.

tracernz, 2015-12-10:
Yeah, this is not a bad idea, especially now with Revo nano it might be even more confusing than ever.

jihlein, 2015-12-14:
I can tackle this one.  Do we want Revolution or OP-Revolution?

mlyle, 2015-12-14:
revolution sounds great :)

tracernz, 2015-12-14:
I vote Revolution, it's consistent with our other target names. This will clash with #166

mlyle, 2015-12-14:
Ah, you may want to build it on top of #166 in the assumption that will merge first

jihlein, 2015-12-18:
I built this on top of #166, it builds cleanly and the GCS runs.  I don't have a Revolution board to test with.

I'll hold off submitting the PR until #166 is merged, then I'll rebase against that and submit the PR.

mlyle, 2015-12-18:
Consider submitting it as a PR against the branch @tracernz has used in
#166.  Then we can test/take them together.  (but after first release,
unfortunately, it looks like)

jihlein, 2015-12-24:
For the GIT challenged, what's the best way to submit this as a PR against tracernz's branch?

mlyle, 2015-12-24:
To start your work, first you need to be based against #166; it sounds like you are, but if you need help with how to rebase on top of that I can.

Click "open a pull request".  For the base fork, choose "tracernz/dronin" and the base branch "revo-movingup"  (You may have to choose "compare across forks") for this to work.

Then for the head fork, pick your branch.

Then, @tracernz will get your change in his PR queue, and can merge it into his branch and maintain it with his PR.

jihlein, 2015-12-24:
Thanks, got it completed.

mlyle, 2015-12-24:
Cool.  I imagine we may not see him today, since it's now Christmas morning in .nz :)

PR #166 (2016-02-04): Revo: Upgrade to ChibiOS, add some missing modules, rename
Not flight tested yet, only brief bench test. Recalibrated the idle count for CPU load. Enabled some modules including flash logging which required re-jigging of the external flash partition table. I also added checks during init and LED panic codes.

[[image]](https://reviewable.io/reviews/d-ronin/dronin/166)

mpl edit: fixes #204 connects #169
closed 2016-02-04 by mlyle