d-ronin/dRonin #205: Setup dRonin Website

mluessi, 2015-12-09:
We should set up a website for the project. The website should contain most of the user facing information:

- General information about the project, i.e., project vision, news feed, etc
- User documentation (probably a WIKI)
- User forum
- ..

I think a good website and documentation is key to acquiring a large user base. IMO, we should try to set this up using as much open source technology as we can. If we need to pay for hosting, we could give users the option to make donations to the project and / or look for sponsors.

Do we have people who have experience with web development who want to volunteer to take a lead on this?

VArcht, 2015-12-09:
I am willing to take the lead on this, more the merrier though.

jhitesma, 2015-12-09:
I can assist with this as well.  It is my day job :)  I can probably provide free hosting for a basic site as long as we stick with things like github for distributing releases and autotown is in the cloud.  Let me know if you want me to see if I can get us a hosting account donated.  We could probably handle hosting a forum as long as it didn't get too busy...but I'm generally wary of forum's on our servers and would rather we look for a hosted solution for that.  The rest shouldn't be a big deal.

dustin, 2015-12-09:
We have free hosting at google already.  This seems like a good start.  We just don't have content and stuff.  We've got tons of flexibility in what we can do from here.  I'm good at functionality, but not as much user design.

We have two google groups already (one for [general use](https://groups.google.com/forum/#!forum/dronin) and one for [development stuff](https://groups.google.com/forum/#!forum/dronin-dev)).  We tentatively agreed to try using these initially since they're hosted, ~zero maintenance and much less crappy than all the other forum stuff I've seen.  Any reason not to?

(my personal bias -- I don't like these stupid forum things that always seem broken, hard to search, and require me to go to a web site instead of giving me the option to just use my email reader)

We've also got the wiki at github.  All wikis are pretty bad at user documentation, but that one has our stuff in it.  We can stitch stuff together a bit better.  Note that github wiki is a git repo and can be machine processed as well.

jhitesma, 2015-12-09:
I'm all for using as much free hosting as possible.  

I'm not a huge fan of a proliferation of forums...I seldom even keep up with RCGroups (though do hang out at FliteTest regularly since they were what got me back into RC.)  

Maintaining our own forum just sounds like a major hassle though.  I've done my time as a forum mod and admin in the past and they're not hats I want to wear again.  Free forum software sucks and commercial stuff does too in it's own horrible ways.  

I'm not a big fan of google groups...and suspect that end users would like some kind of forum presence.  I'm willing to help with promoting/helping on existing forums but I'm not really up for the extra work of maintaining our own.  So I'm in favor of sticking with the groups, even though I'm not huge on them, and helping with representing us in threads in existing forums.  Maybe in a year or two if the project gets a ton of end users our own forum would make sense...but for now I agree it's not really needed.

The wiki needs a lot of work on content - both removing junk and adding new/better stuff.  I'm kind of picking at that a bit but things are shifting so much right now it's not a high priority for me until things stabilize a bit more.  I think maintaining it on github but then cloning it to the site and processing it into something more attractive is a very good idea.  

One of the big complaints I hear about Tau from people I get to try it is that the presentation is too dense and directed at developers rather than users.  The wiki is a perfect example.  End users don't want that.  They want something simpler and prettier.  I really think a good website with content pulled from the wiki but formatter better could go a long way towards helping adoption.

dustin, 2015-12-09:
> One of the big complaints I hear about Tau from people I get to try it is that the presentation is too dense and directed at developers rather than users. The wiki is a perfect example. End users [...]

I agree with all of this.  I've only seen wikis be really great when they were very strictly maintained -- effectively being more of a CMS than a wiki.  Most wikis are just full of outdated/incorrect/incomplete logs of what a person did around the time something started working.

However, the alternative around here is a forum, which is the same thing, but without being able to collapse it into a coherent work.  So it's a *little* better, but not what to strive for.

tracernz, 2015-12-10:
We should look to pull releases info from Github API etc. like I did for the Tau website. Try to keep the extra maintenance required just for the website to a minimum by pulling the information from elsewhere or it will end up going stale.

dustin, 2015-12-10:
Sure.  I can lug these things around/copy them into blob store or
whatever.  We should figure out what we want in the web site.  I've got a
high level idea, but I will not make anything pretty -- almost on purpose.

jhitesma, 2015-12-11:
With @varcht saying he's burned out and doesn't want to tackle the website I sent a message to our top designer asking what it would take for him to do a basic design for us.  We'd have to handle content but he'd give us a "template" or base design and set of CSS rules for content to work with.  He's only asking for a growler of my homebrew in return and I'm more than willing to donate that for the project.  

I'd need to give him some basic info like what kind of things we'd want in a menu and what kind of look we're going for.  I could try to get him on github but he's a designer not a coder so I'm not sure how well he'd cope with that

dustin, 2015-12-11:
I didn't know there were still designers that didn't code.  :)

Ideally, we've got sensible use of css and stuff we can apply to templates and what not.  We've got basic outer shell we can inject stuff into, but there's little content there.  However, I think we could pretty easily sketch out a map of what we need.

We need a home page that gets you familiar enough quickly (e.g., this is awesome flight control software built by people too smart for you to talk to (example copy, find something better)).  

We've got a few data sources to make it look interesting (recent development, bug/commit info, etc...).

More detailed about/FAQ page.  Why does this exist?  Why isn't it Tau Labs/OpenPilot/librepilot.  But all I know is cleanflight, which is the best, why would I use something else?

High level docs for hardware and software concepts (and/or lob over to wiki, but ideally the bulk of docs isn't in wiki format).

I'm sure people can think of other things, but we should probably have an ideal outline at least in parallel with some high level UX.

jhitesma, 2015-12-11:
He's picked up a little bit of JS but his indenting gives me headaches, I'm still working on teaching him git.  But he's a really great artist and does HTML/CSS very well.

I agree 100% with the basic ideas you're outlining and agree we should get a solid outline of just what we want.  I just also think it's important that we have a good image if we're going to try and attract a wider user base.  Almost everyone I've tried to turn on to Tau has been turned off by the website and quickly decided it was for PhD's and not end users, I just don't want to see us make that same mistake.

Varcht's given us some really great artwork and logo and we should let them shine with equally good design that's both functional and aesthetically pleasing.

The recent info from git is a great idea - I'm always a fan of fresh content to keep users returning to the site.  But I don't think it should be as dominant as it was on Tau's.  But I think our main focus should be on the About/FAQ/Docs being presented in a user friendly way.  I'm still torn as to whether we should just reformat data from the wiki...or do something separate, I see too many pros/cons to both approaches.  I think the wiki is valuable for developers.  But it would be nice to create a more condensed, attractive and user friendly end user documentation on the website.

hernick-qc, 2015-12-11:
We could use the github d-ronin.github.io wiki for planning the site. It's great for working on content which can then be incorporated into the real site. https://github.com/d-ronin/d-ronin.github.io/wiki

mluessi, 2015-12-11:
Is it possible to pull the WIKI content from github and then statically "compile it to HTML"? This would allow people to add content to the WIKI on gh and at the same time it would look pretty to users. Another project I have been working on uses [Sphinx](http://sphinx-doc.org/) which is very Python centric, but maybe similar tools exist for WIKIs.

hernick-qc, 2015-12-11:
It is certainly doable, using for example a markdown compiler like marked (https://github.com/chjj/marked) to generate HTML. Of course, it might not be the best way to do things. If the dRonin web site is based on a CMS with built-in staging support, using the wiki as a staging area might just be a pain nobody wants to deal with.

I see the web site planning wiki more as a temporary measure, to develop the web site content while the site isn't online yet.

dustin, 2015-12-11:
We could do something like that -- potentially even tagging content for
inclusion in such a thing.  It's a good idea to explore.

hernick-qc, 2015-12-11:
Ok, there's been proposals on #dronin to use either of the three following static web site generators. Clean and simple, and they support markdown too.

Jekyll https://jekyllrb.com/
Hugo http://gohugo.io/
Middleman https://middlemanapp.com/

Certainly less work and trouble than a full-fledged CMS.

dustin, 2015-12-11:
The huge advantage of jekyll is you don't need any software to use it.

closed 2016-01-15 by mlyle

tracernz, 2016-01-15:
The website looks great now. Thanks to all involved! :-D