d-ronin/dRonin #206: Create Flight Test Team

mluessi, 2015-12-09:
It would be good to have a dedicated group of people with a variety of targets / UAVs who would be willing to test experimental code and report back their findings. Flight test team members don't necessarily have to be developers, but should be able to pull a branch from github and build it (or maybe they can just download a jenkins artifact).

dustin, 2015-12-09:
I've been doing most of my testing on ~180s with sparky, naze, and Brain.  I've got a qav250 with a naze32 that needs to be converted.  I've also got a ~550 tri with a cc3d and an ancient TL build on it, but LoS only (i.e., I never fly it -- need to do something about that).

We could do this with more structure (need X feature tested, limiting scope a lot), but right now, the users are so few, it's probably easiest just to have a mailing list for them.

jhitesma, 2015-12-10:
I have three builds I'm willing (and have been) using for testing - and a 4th I'm willing to use once things are at a release stage.

1) 250 emax nighthawk - basher quad, 1806 motors, "simon series" emax esc's naze32  
2) 280 emax nighthawk pro - really enjoy flying it but was donated to me so I don't mind doing stupid things with it.  2204's with funky emax all in one board (naze32 clone with silabs esc's I've reflashed to blheli 14.2)
3) Homemade 380 sized wood frame test mule.  Was a flyingF4 but putting a sparky on it.  2212 1400kv motors and 20a blue series ESC's with blheli 14.2 swinging 8" props.

The first two have successfully flown dRonin and the new autotune.  The third I'm still setting up.

I also have a mini hex (2204 2300kv motors with 18a kiss esc's and a brain board) but I can't afford to do anything too crazy with it so I'm only willing to do ground testing or final release testing with it ;)

The 250 and the 380 I'm most willing to experiment with and put in harms way.

mlyle, 2017-02-06:
Doesn't lend itself to tracking in an issue.

closed 2017-02-06 by mlyle