d-ronin/dRonin #207: Throttle based PID Gain Scheduling

mluessi, 2015-12-09:
We currently have TPA, which can reduce the PID gains based on throttle. However, this is kind of a hack and should ideally be handled by autotune. I haven't studied the autotune derivation in detail, but I wonder if we can collect the throttle position during autotune (so we know the stick position for hover) and then extrapolate to determine PID gains at different throttle positions (using an optimization procedure in the GCS). We would then need some way to store this efficiently and generate gains during flight. We could use a lookup table with linear interpolation, or maybe a low order polynomial for this.

tracernz, 2015-12-09:
We can just use the actuator scaling to achieve this, if we could tune it automatically that would be superb.

mluessi, 2015-12-09:
@tracernz yes, I guess if the actuator curve fitting makes things perfectly linear, the throttle PID scaling should be linear. Unless there are some other non-linear effects..

mlyle, 2015-12-10:
@tracernz  So technically the accelerometer gives us a pretty good hint about instantaneous actuator performance...

ufoDziner, 2016-09-19:
When this feature is implemented, it would be great if the hangtime throttle percentage was set based on the results.

mlyle, 2018-03-13:
Upon looking at the data, this isn't practical.. and we're doing way better with regards to this anyways. opening a follow-on issue-- superseded #2114

closed 2018-03-13 by mlyle