d-ronin/dRonin #213: packaging: LICENSE.TXT needs to ship

mlyle, 2015-12-10:


mlyle, 2015-12-11:
@hernick-qc has done some great work cleaning up the license.txt.  We need to ensure appropriate copyright notices for shipped code get in.  There is also conflict with #151 because that particular license requires we ship it.

[Also for consideration when I get to this-- should I include the individual license files that go into software- GPL/LGPL?  TBD.  In the case of Berkeley 3 clause like the fastapprox I need to]

hernick-qc, 2015-12-11:
It's not necessary to include the actual license files, a link to the original license text from a license provider (gnu.org, creativecommons...) is enough.

Also, there are 511 files already in the code with 3-clause BSD licenses (keyword: Redistributions). They're fine where they are, in the actual code files. The general disclaimer in LICENSE.txt covers that.

mlyle, 2015-12-11:
```
//     * Redistributions in binary form must reproduce the above
// copyright notice, this list of conditions and the following disclaimer
// in the documentation and/or other materials provided with the
// distribution.
```

To me this means (and legal counsel has advised me in the past) we need to gather it up a file with all of this stuff and ship it together.

hernick-qc, 2015-12-11:
The 3-clause BSD is very permissive... A link to the github shipped with the binaries is documentation and/or material provided with the distribution.

Now, if this were a closed-source project it would be necessary to ship the license with the binaries. But since full access to the source code is provided with the distribution... this satisfies the five-hundred-plus 3-clause BSD licenses used in dRonin.

Almost everybody else does just that. No need to spend dozens of hours trying to put together the most impressive license file package in the world.

hernick-qc, 2015-12-12:
The Windows installer should show LICENSE.txt

closed 2015-12-19 by mlyle