d-ronin/dRonin #216: Packagework1 Cleanup: Linux

mlyle, 2015-12-10:
Cleanup from #177

* Linux deb packaging doesn't work
* Linux deb menus/icons/etc are mismatched/misbranded

PTDreamer, 2015-12-10:
Can you clarify if you want me to stand down or can I work on this and open a PR against your branch?

mlyle, 2015-12-10:
If you want to handle this one, I'd appreciate it.  My branch has merged, so you can open it against the main project.

mlyle, 2015-12-10:
At the same time, I realize I've left a bit of a mess and I don't expect you to clean it up-- only if you want to work on this.

PTDreamer, 2015-12-10:
I'll take it :)

fujin, 2015-12-10:
It would be nice for this to not be Debian specific.

mlyle, 2015-12-10:
@fujin there's a package_all_compress if you just want the bins in a dir, and there's tools to convert debs to other packages.  I don't think we need to complicate our lives further by maintaining many variant packaging formats.

fujin, 2015-12-11:
@mlyle that's not what I was suggesting - I meant not maintaining the Debian-specific build capabilities. Let a debian package maintain that in universe or whatever if they like. Produce tars, sigs, shas. Whatever though :cool:

tracernz, 2015-12-11:
I vote we maintain the Debian packaging. It's very popular and we already steer people towards Debian in all our documentation. There will still be a zip package suitable for non-Debian dists.

mlyle, 2015-12-16:
Fixed (at least mostly) by #253 .

closed 2015-12-16 by mlyle